EmuELEC/EmuELEC #613: GT King S922x Rev b

gonzaspt, 2021-05-17:
**Describe the bug**
GT King Amogic S922x rev b always freezing/crashing and restarting.
tried with both DTBs (base and rev a), same behaviour.

**To Reproduce**
Found so far crashes during:
- Enabling/disabling Wi-fi
- Restarting or shutdown
- Opening/Closing retroarch 
- On saving metadata

**Expected behavior**
Do not crash

**EmuELEC version:**
 - Version 4.1

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S922x rev b
 - Brand: Beelink
 - Model GT king
 - Version: revb

**Additional context**
Either in ES or RA, it shows only 4 cores instead of 6 cores

shantigilbert, 2021-05-17:
Without logs its impossible to even start guessing what might be the issue, random crashes/hangs are not typical, so its either something on your device end (bad SD, bad HW) or a setting you changed that is causing this. 

4 Cores is normal, EmuELEC disables the small cores so that it always uses the bigger faster ones.

gonzaspt, 2021-05-17:
I think this is a recent model (the rev b). Perhaps a new dtb is needed for rev b?
I've also tried the dtb.img that came with the Coreelect image in the 8Gb card with the box, and I think it got better, but still crashes. I'll do some more testing and get the logs.

shantigilbert, 2021-05-17:
Does CoreELEC crash as well?

gonzaspt, 2021-05-17:
No, coreELEC runs just fine.

shantigilbert, 2021-05-17:
Have you tried running something that pushes the CPU ? 4k video, or an emulator addon ?

gonzaspt, 2021-05-17:
Yes, in CoreELEC, no crashes at all

shantigilbert, 2021-05-17:
Well logs are probably what we need then, as I have a Rev B and it works fine, no issues nor crashes

gonzaspt, 2021-05-17:
what dtb do you use for your rev b?

shantigilbert, 2021-05-17:
g12b_s922x_beelink_gt_king

gonzaspt, 2021-05-17:
ok, thanks

shantigilbert, 2021-05-17:
I am going to close this as its not really an EmuELEC issue, feel free to continue this on the forum (https://emuelec.org) or discord server. Also feel free to reopen it if it turns out that in fact is a reproducible issue/bug on EmuELEC.

closed 2021-05-17 by shantigilbert