EmuELEC/EmuELEC #614: Weird bug where the screen is black on launching a game

Cee123, 2021-05-18:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
I'm getting this weird issue where, whenever I launch a game on EmuElec (Odroid Go Advance), it goes play but the sound is playing and I have to enter to Retroarch menu and then exit it again for the game to appear.

**To Reproduce**
I'm not sure how to reproduce this. Just wondering if anyone else has the same issue.

**Expected behavior**
The game should just display, however, it doesn't. You have to access the retroarch menu first.

**Screenshots**
-

**EmuELEC version:**
4.x (tried 4.1 and 4.2) test versions

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Odroid Go Advance

**Additional context**
I updated using the tar.gz file, instead of a complete reflash of the SD card. I really didn't want to go through all of the copying stuff over etc again as last time it took over an hour.

shantigilbert, 2021-05-22:
I no longer have a OGA v1.0 so I cannot test this, but on all the other RK3326 devices I have with EmuELEC I cannot reproduce.

Cee123, 2021-05-23:
> I no longer have a OGA v1.0 so I cannot test this, but on all the other RK3326 devices I have with EmuELEC I cannot reproduce.

Oh ok, no worries. It could be something simple to do with retroarch maybe. I'll try different settings see if I can fix it.

shantigilbert, 2021-05-23:
Let me know if you find a way to reproduce this

shantigilbert, 2021-05-31:
No info for 8 days, closing

closed 2021-05-31 by shantigilbert

Cee123, 2021-07-09:
@shantigilbert I just managed to fix this issue today. It was simple a fix as changing the video driver from gl to glcore.