EmuELEC/EmuELEC #616: Game options are missing in the latest test release

Cee123, 2021-05-20:
**Describe the bug**
There are no more game options. Inability to change the emulator or settings per game.

**To Reproduce**
Update from version 4.0 to 4.2 Test release, via tar.gz file.

**Expected behavior**
Not sure why the game options have disappeared.

**Screenshots**
[image]

**EmuELEC version:**
v4.2-TEST-05142021

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Odroid Go Advance

**Additional context**
Things I've tried
- Resetting/restoring all the configs to default
- Scrolling down past Advanced System Options (there are no further options listed)
Things I haven't tried
- Reflashing the entire image

shantigilbert, 2021-05-20:
Consider using the forum for these kind of questions, anyways, press and hold the A button

closed 2021-05-20 by shantigilbert

Cee123, 2021-05-20:
> 
> 
> Consider using the forum for these kind of questions, anyways, press and hold the A button

Oh I see, thanks. It had just been moved. Ok yeah that works now.