EmuELEC/EmuELEC #617: ps3 dualshock bluetooth mapping problem

spartanj99, 2021-05-22:
i have problem with original ps3 dualshock,bluetooth button mapping is messed up only in native apps like scummvm/ppsspp or game ports like sonic1,sonic2...i could check in scummvm some buttons work, like confirm with select button(instead of x),analog works,left shoulder not working...
when connected to usb, to work as wired gamepad,button mapping is fine and everything works,when unplugged different mapping. this problem is only for native emulators,ports,in emulationstation and retroarch everything works fine...
had this same problem with 4.0 version also...

**EmuELEC version:**
 - Version 4.1

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: s905
 - Brand: minimx

shantigilbert, 2021-05-22:
Please use the forum or discord for these kind of questions. PS3 gamepads are notorious for having all sorts of problems on EmuELEC, there is no know solution, but it is not considered a bug since its so random and it does not happen to everyone, unless someone can do a deep debug the only solution is to change gamepad.

closed 2021-05-22 by shantigilbert

spartanj99, 2021-05-22:
ok,thx