EmuELEC/EmuELEC #626: FAILURE: scripts/install gptokeyb has failed @4.1 code

liaohouwei2020, 2021-05-28:
My PC version Ubuntu 20.04.2 LTS,
Execution under the project： PROJECT=Amlogic ARCH=aarch64 DISTRO=EmuELEC make image
An error occurred as follows：
BUILD      gptokeyb (target)
    TOOLCHAIN      make
Executing (target): make 
make[1]: Entering directory '/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/gptokeyb-30b4e9d526ad87bce078bfc2f585a76aa70b96c6'
/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -W -Wall `/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin/sdl2-config --cflags` -I/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/include/libevdev-1.0 "gptokeyb.cpp" -o gptokeyb -levdev `/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin/sdl2-config --libs`
/bin/sh: 1: /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin/sdl2-config: not found
/bin/sh: 1: /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin/sdl2-config: not found
gptokeyb.cpp:46:10: fatal error: libevdev-1.0/libevdev/libevdev-uinput.h: No such file or directory
 #include <libevdev-1.0/libevdev/libevdev-uinput.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:10: all] Error 1
make[1]: Leaving directory '/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/gptokeyb-30b4e9d526ad87bce078bfc2f585a76aa70b96c6'
FAILURE: scripts/install gptokeyb has failed!
[296/460] [FAIL] install gptokeyb

shantigilbert, 2021-05-28:
Check the dev branch for the fix

leoliao2021, 2021-05-28:
git checkout v4.1 ？

leoliao2021, 2021-05-28:
Hi shantigilbert:
Do you mean this?
leoliao@leoliao-virtual-machine:~/project/EmuELEC$ git branch 
  dev
* master
leoliao@leoliao-virtual-machine:~/project/EmuELEC$ git checkout master
Already on 'master'
Your branch is up to date with 'origin/master'.
leoliao@leoliao-virtual-machine:~/project/EmuELEC$ git pull  origin master

leoliao2021, 2021-05-28:
I'm following the rules：
sudo apt update && sudo apt upgrade
sudo apt-get install gcc make git unzip wget xz-utils libsdl2-dev libsdl2-mixer-dev libfreeimage-dev libfreetype6-dev libcurl4-openssl-dev rapidjson-dev libasound2-dev libgl1-mesa-dev build-essential libboost-all-dev cmake fonts-droid-fallback libvlc-dev libvlccore-dev vlc-bin texinfo premake4 golang libssl-dev curl patchelf xmlstarlet default-jre xsltproc
git clone https://github.com/shantigilbert/EmuELEC.git EmuELEC    
cd EmuELEC  
git checkout master  
PROJECT=Amlogic ARCH=aarch64 DISTRO=EmuELEC make image

shantigilbert, 2021-05-28:
https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/tools/sysutils/gptokeyb/package.mk

leoliao2021, 2021-05-28:
The problem should be that there is no libevdev-1.0 in my ubuntu system. How do I install libevdev package?

shantigilbert, 2021-05-28:
Update the package! copy the data from dev, I already linked it!

leoliao2021, 2021-05-28:
Thanks! shantigilbert
This package has been compiled successfully.

closed 2021-05-28 by leoliao2021