EmuELEC/EmuELEC #627: build install dosbox-svn failed#4.1

leoliao2021, 2021-05-28:
BUILD      dosbox-svn (target)
    TOOLCHAIN      make
make[1]: Entering directory '/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/dosbox-svn-97425232734636594d615f900b5971498451d794/libretro'
/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -D__LIBRETRO__ -MMD -fPIC  -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION=\"" 97425232"\" -DSVN_VERSION=\"" r4393"\" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags`   -c -o ../libretro/libretro_gfx.o ../libretro/libretro_gfx.cpp
/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -D__LIBRETRO__ -MMD -fPIC  -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION=\"" 97425232"\" -DSVN_VERSION=\"" r4393"\" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags`   -c -o ../libretro/libretro_mapper.o ../libretro/libretro_mapper.cpp
/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -D__LIBRETRO__ -MMD -fPIC  -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION=\"" 97425232"\" -DSVN_VERSION=\"" r4393"\" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags`   -c -o ../libretro/libretro.o ../libretro/libretro.cpp
/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -D__LIBRETRO__ -MMD -fPIC  -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION=\"" 97425232"\" -DSVN_VERSION=\"" r4393"\" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags`   -c -o ../src/cpu/callback.o ../src/cpu/callback.cpp
cc1plus: fatal error: retro_endianness.h: No such file or directory
compilation terminated.
cc1plus: fatal error: retro_endianness.h: No such file or directory
compilation terminated.
cc1plus: fatal error: retro_endianness.h: No such file or directory
compilation terminated.
make[1]: *** [<builtin>: ../libretro/libretro_gfx.o] Error 1
make[1]: *** Waiting for unfinished jobs....
cc1plus: fatal error: retro_endianness.h: No such file or directory
compilation terminated.
make[1]: *** [<builtin>: ../libretro/libretro.o] Error 1
make[1]: *** [<builtin>: ../src/cpu/callback.o] Error 1
make[1]: *** [<builtin>: ../libretro/libretro_mapper.o] Error 1
make[1]: Leaving directory '/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/dosbox-svn-97425232734636594d615f900b5971498451d794/libretro'
FAILURE: scripts/install dosbox-svn has failed!
[321/460] [FAIL] install dosbox-svn

The following logs for this failure are available:
  stdout: /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/.threads/logs/323/stdout
  stderr: /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/.threads/logs/323/stderr

parallel: This job failed:
package_worker 2 323 460 'install dosbox-svn'
Parallel build failure - see log for details. Time of failure: Fri May 28 13:30:57 CST 2021
make: *** [Makefile:12: image] Error 1

error info:retro_endianness.h: No such file or directory
Also need to update data? How to solve this problem?
thanks!

shantigilbert, 2021-05-28:
Update the packages that fail from dev, its the same procedure.

closed 2021-05-28 by shantigilbert

leoliao2021, 2021-05-28:
I update package and copy GitHub file into my project.(packages/sx05re/libretro/dosbox-svn/package.mk)
but why also report this errors？

zongki, 2023-02-03:
> BUILD dosbox-svn (target) TOOLCHAIN make make[1]: Entering directory '/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/dosbox-svn-97425232734636594d615f900b5971498451d794/libretro' /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3 -D__LIBRETRO__ -MMD -fPIC -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION="" 97425232"" -DSVN_VERSION="" r4393"" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags` -c -o ../libretro/libretro_gfx.o ../libretro/libretro_gfx.cpp /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3 -D__LIBRETRO__ -MMD -fPIC -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION="" 97425232"" -DSVN_VERSION="" r4393"" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags` -c -o ../libretro/libretro_mapper.o ../libretro/libretro_mapper.cpp /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3 -D__LIBRETRO__ -MMD -fPIC -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION="" 97425232"" -DSVN_VERSION="" r4393"" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags` -c -o ../libretro/libretro.o ../libretro/libretro.cpp /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3 -D__LIBRETRO__ -MMD -fPIC -I../src -I../libretro -I../libretro/deps/common/include -I../include -I../src/midi/munt -include retro_endianness.h -include dynarec.h -DC_HAVE_MPROTECT="1" -O3 -DNDEBUG -DGIT_VERSION="" 97425232"" -DSVN_VERSION="" r4393"" `pkg-config --cflags sdl` -DC_IPX `sdl-config --cflags` -c -o ../src/cpu/callback.o ../src/cpu/callback.cpp cc1plus: fatal error: retro_endianness.h: No such file or directory compilation terminated. cc1plus: fatal error: retro_endianness.h: No such file or directory compilation terminated. cc1plus: fatal error: retro_endianness.h: No such file or directory compilation terminated. make[1]: *** [: ../libretro/libretro_gfx.o] Error 1 make[1]: *** Waiting for unfinished jobs.... cc1plus: fatal error: retro_endianness.h: No such file or directory compilation terminated. make[1]: *** [: ../libretro/libretro.o] Error 1 make[1]: *** [: ../src/cpu/callback.o] Error 1 make[1]: *** [: ../libretro/libretro_mapper.o] Error 1 make[1]: Leaving directory '/home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/dosbox-svn-97425232734636594d615f900b5971498451d794/libretro' FAILURE: scripts/install dosbox-svn has failed! [321/460] [FAIL] install dosbox-svn
> 
> The following logs for this failure are available: stdout: /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/.threads/logs/323/stdout stderr: /home/leoliao/project/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.1/.threads/logs/323/stderr
> 
> parallel: This job failed: package_worker 2 323 460 'install dosbox-svn' Parallel build failure - see log for details. Time of failure: Fri May 28 13:30:57 CST 2021 make: *** [Makefile:12: image] Error 1
> 
> error info:retro_endianness.h: No such file or directory Also need to update data? How to solve this problem? thanks!

you need to update the submodule "libretro/deps/common" to obtain the missing files.