EmuELEC/EmuELEC #633: Gameforce CHI - 4.1 Stable - Retroachievements not working

matteozel, 2021-06-05:
Ciao,

I'm having a strange issue with retroachivements in 4.1 on the gameforce chi.
I can login in emulationstation with my account, I can see the achievements list in emulationstation but in retroarch retroachievements is not working properly.

In retroarch settings I can see username and password (both are correct) but the "achievements" menu in the quick menu says that I'm not logged in

[image]

I've tried to re-enter username and password in retroarch, also I've re-flashed from scratch 41 but the issue still occur.
With 4.1 on OGS there is no issue

(same rom, same emulator) 

[image]

shantigilbert, 2021-06-05:
Can you try latest 4.2-test?

shantigilbert, 2021-06-05:
[image]
I am using the GF with 4.2-test and this seems to work fine

matteozel, 2021-06-06:
Same issue after the update to the latest 4.2 test, I've done some other test and this is what I've understand:
- problem it's related to SNES9X emulator (I've tried the different snes9x cores)
- problem is on 4.1 and 4.2 test as well
- launching another emulator/rom (eg: Genesis Plus GX + Sonic 1), retroachievements is working as expected:
[image]
- launching again snes9x the message is different but the issue is the same (no achievements):
[image]

Screens are from 4.1 but it's the same on 4.2.
As you can see from this:
[image]
it seems that with snes9x core retroarch is not able to reach retroachievements at all. The snes rom is the same as on the first post so I'm sure that the rom is supported.

shantigilbert, 2021-06-06:
I have no idea why this is happening, and I cannot reproduce it, Using Snes9x on my Chi, achievements work as usual 
[image]

shantigilbert, 2021-06-06:
[image]
[image]

here another 2 screenshots one before and one after unlocking an achievement

matteozel, 2021-06-07:
Thanks for the feedback @shantigilbert, probably I'm making some dumb mistake (like wrong rom or whatever), I'm closing this one and I will investigate on my own :)

closed 2021-06-07 by matteozel

shantigilbert, 2021-06-07:
Let me know if you find something that I am missing.