EmuELEC/EmuELEC #634: Amiberry broken on 4.2 latest test

0kti, 2021-06-07:
When I choose game (lha format) it's starts, but it goes back to emulation station.
On stable 4.1 was ok.

shantigilbert, 2021-06-07:
Can you please provide logs?

0kti, 2021-06-07:
`EmuELEC Amiberry Log
Trying to boot this game: /storage/roms/amiga/Aladdin.lha
Loading a WHDLoad (.lha) file...
Loading Amiga 500 config.
Gamepad used GO-Advance Gamepad (rev 1.1)
`

`EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.2-TEST-06062021 (OdroidGoAdvance.aarch64)
PLATFORM: amiga
ROM NAME: /storage/roms/amiga/Aladdin.lha
BASE ROM NAME: Aladdin.lha
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/amiga/Aladdin.lha
2nd Argument: -Pamiga
3rd Argument: --core=AMIBERRY
4th Argument: --emulator=AMIBERRY
Full arguments: /storage/roms/amiga/Aladdin.lha -Pamiga --core=AMIBERRY --emulator=AMIBERRY --controllers=-p1index 0 -p1guid 190000004b4800000010000001010000 
Run Command is:
/usr/bin/bash /usr/bin/amiberry.start /storage/roms/amiga/Aladdin.lha
Emulator Output is:
Segmentation fault
exit 0
`

shantigilbert, 2021-06-07:
Thank you, Is this only for lha?

0kti, 2021-06-07:
I Didn't check adf files.
Maybe Amiberry 4.xx is broken that's why all rollback to 3.3 version

shantigilbert, 2021-06-07:
For the time being I am going to revert Amiberry until I can find a solution.

Commit: ccee0f3f0719495bb7c03b14ccc64fcaabce49a4

closed 2021-06-18 by shantigilbert