EmuELEC/EmuELEC #638: Rotation for Gameforce-Chi

qbertaddict, 2021-06-12:
Rotation on the gameforce chi is not currently available like it is on the oga.

I've tested using the rotate option in the current cores and the resolution on arcade games looks better when rotated and it uses the full screen. 

Wonderswan, Arcade, Mame, Fba, and Vectrex would all work with this feature and it looks much better than they currently do.

Langerz82, 2022-06-07:
I know this is old but if someone has this issue still can the post a screenshot of how it currently appears and what they would like to see change.

djibux, 2023-12-24:
This can be done with minor tinkering, at least on libretro cores.

Open retroarch menu and go to `Core Options > Video > Tate Mode`.
Then go to `Controls > Port 1 Controls` and map the right joystick as follows:
* Right analog up: Right
* Right analog down: Left
* Right analog left: Up
* Right analog right: Down

You can save the remap using `Controls > Manage Remap File` and save for the specific game.

It’s quite enjoyable to get a full screen display… but holding the console vertically feels awkward and can be painful for long gaming sessions.

Enjoy!

[image]

Langerz82, 2024-07-10:
It would be nice in retrorun if we have arguments to rotate the screen or the controllers, I could create a patch but I think someone was working on this feature. That was in ES we could make options to rotate the view or controllers per system or per game.

Langerz82, 2024-09-01:
This PR will enable users to rotate the game screen for retroarch only. So it is a partial fix for this issue, see:
https://github.com/EmuELEC/EmuELEC/pull/1388

Will be release in EE V4.9 most likely.

Langerz82, 2024-09-02:
https://github.com/EmuELEC/EmuELEC/pull/1393

Should fix the issue and allow retrorun to work in Tate Mode.

closed 2024-09-02 by Langerz82