EmuELEC/EmuELEC #643: Feature - For N64 option to launch Mupen64plus natively without Retroarch.

Langerz82, 2021-06-20:
**Is your feature request related to a problem? Please describe.**
Retroarchs n64 emulator is slower and has some graphical glitches with games because of outdated code.

**Describe the solution you'd like**
An N64 option to launch Mupen64plus natively without retroarch.

**Describe alternatives you've considered**
None

**Additional context**
It would be super cool. Allot of GPU's are lacking on small devices and we need all the FPS and stability we can get!
If someone could point me in the right direction, maybe I could help with it.
The issue I do see is different controller configurations might have to be added in. Retroarch is a great wrapper but adds a bit of overhead.

shantigilbert, 2021-06-21:
> Retroarchs n64 emulator is slower and has some graphical glitches with games because of outdated code.
Source? 

There are many issues on why this is not included, at some point it was included but later removed, I have no idea what device you have since you left that out, but on Amlogic devices the standalone emulators runs really bad, I am not 100% sure it's settings related, but I have no real time to spend on testing a N64 emulator.  I spent a lot of time trying and for me its just not worth it, not saying its impossible, just not something I was successful. 

Standalone emulator has updated code, that is correct, but its also code to make the emulator run more accurately, which means its slower on low-end devices. 

So if someone wants to take the time to implement it and make it run as you claim it should, PRs are welcome

Langerz82, 2021-06-27:
Hey,

Sorry I'm using a Beelink GT King Pro.

I disabled the shaders in retroarch overrides, and enabled anti-aliasing and im getting a decent graphics render at ~30FPS for 3D. I also set the overclock to 2200 and Framerate is set to Fullspeed.

For some reason my cheap chinese PS3 controller knock-offs were switching the analog values from left-to-right and right-to-left  so I had to overwrite some settings in ".config/retroarch/config/Mupen64Plus-Next GLES2/Mupen64Plus-Next GLES2.cfg"

'''
input_libretro_device_p1 = "1"
input_libretro_device_p2 = "1"
input_libretro_device_p3 = "1"
input_libretro_device_p4 = "1"
input_player1_analog_dpad_mode = "0"
input_player2_analog_dpad_mode = "0"
input_player3_analog_dpad_mode = "0"
input_player4_analog_dpad_mode = "0"
input_player1_l_x_minus_axis = "-0"
input_player1_l_x_plus_axis = "+0"
input_player1_l_y_minus_axis = "-1"
input_player1_l_y_plus_axis = "+1"
input_player2_l_x_minus_axis = "-0"
input_player2_l_x_plus_axis = "+0"
input_player2_l_y_minus_axis = "-1"
input_player2_l_y_plus_axis = "+1"
input_player3_l_x_minus_axis = "-0"
input_player3_l_x_plus_axis = "+0"
input_player3_l_y_minus_axis = "-1"
input_player3_l_y_plus_axis = "+1"
input_player4_l_x_minus_axis = "-0"
input_player4_l_x_plus_axis = "+0"
input_player4_l_y_minus_axis = "-1"
input_player4_l_y_plus_axis = "+1"
input_player1_r_x_minus_axis = "-2"
input_player1_r_x_plus_axis = "+2"
input_player1_r_y_minus_axis = "-3"
input_player1_r_y_plus_axis = "+3"
input_player2_r_x_minus_axis = "-2"
input_player2_r_x_plus_axis = "+2"
input_player2_r_y_minus_axis = "-3"
input_player2_r_y_plus_axis = "+3"
input_player3_r_x_minus_axis = "-2"
input_player3_r_x_plus_axis = "+2"
input_player3_r_y_minus_axis = "-3"
input_player3_r_y_plus_axis = "+3"
input_player4_r_x_minus_axis = "-2"
input_player4_r_x_plus_axis = "+2"
input_player4_r_y_minus_axis = "-3"
input_player4_r_y_plus_axis = "+3"
'''
Hope this may be some use if anyone has issues too.

shantigilbert, 2021-06-27:
> Also I'd like to download the parallel64 core. What should be the core updater url?

You should not, cores are compiled specifically for EmuELEC and in most cases should not be replaced as they will give you issues. But the one you need for parallel is the 32bit ARM version, and it has to be renamed accordingly.

closed 2021-06-27 by shantigilbert

Langerz82, 2021-06-28:
Can we get the following emulators in next version if they havent been included?
parallel64
fba-cps1
fba-cps2

I dont think I could launch the Capcom emulators even though there in cps1 and cps2 folders.