EmuELEC/EmuELEC #646: Failed to compile, please help!!!

earth2004, 2021-06-23:
Just wanted to compile a img but stuck at [374/463] [FAIL] install spring
Cannot get spring sources : https://github.com/valadaa48/spring-libretro/archive/443d2a83584a6ea26cba797a88b5f4dae2258402.tar.gz
Try later!

Any help/hints will be much appreciated.

shantigilbert, 2021-06-23:
just remove that package as dependency in distributions/EmuELEC/options. Spring is not been used anyways

closed 2021-06-23 by shantigilbert

shantigilbert, 2021-06-23:
or just pull latest commit

earth2004, 2021-06-23:
Thank you for the prompt response, and it works after I delete spring. But
another error showing

GET      duckstation (archive)
--2021-06-23 06:21:56--
https://www.duckstation.org/libretro/duckstation_libretro_linux_aarch64.zip
Resolving www.duckstation.org (www.duckstation.org)... 45.63.29.71
Connecting to www.duckstation.org (www.duckstation.org)|45.63.29.71|:443...
connected.
HTTP request sent, awaiting response... 200 OK
Length: 4037418 (3.8M) [application/zip]
Saving to:
'/home/louis/EmuELEC/sources/duckstation/duckstation-8b234c06.zip'

/home/louis/EmuELEC 100%[===================>]   3.85M  2.76MB/s    in 1.4s

2021-06-23 06:21:59 (2.76 MB/s) -
'/home/louis/EmuELEC/sources/duckstation/duckstation-8b234c06.zip' saved
[4037418/4037418]

    WARNING      Incorrect checksum calculated on downloaded file: got
0ab866981b13150c6e67336317eed002f26e5b26d2955c73d8eb3c6d67cc58fb wanted
3a986bc77837e92a116413956c6863b876effb018dbfd5da962d26ebd0f1c4b1
--2021-06-23 06:21:59--
http://sources.libreelec.tv/mirror/duckstation/duckstation-8b234c06.zip
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80...
connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location:
https://sources.libreelec.tv/mirror/duckstation/duckstation-8b234c06.zip
[following]
--2021-06-23 06:22:01--
https://sources.libreelec.tv/mirror/duckstation/duckstation-8b234c06.zip
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443...
connected.
HTTP request sent, awaiting response... 404 Not Found
2021-06-23 06:22:03 ERROR 404: Not Found.

--2021-06-23 06:22:03--
https://www.duckstation.org/libretro/duckstation_libretro_linux_aarch64.zip
Resolving www.duckstation.org (www.duckstation.org)... 45.63.29.71
Connecting to www.duckstation.org (www.duckstation.org)|45.63.29.71|:443...
connected.
HTTP request sent, awaiting response... 200 OK
Length: 4037418 (3.8M) [application/zip]
Saving to:
'/home/louis/EmuELEC/sources/duckstation/duckstation-8b234c06.zip'

/home/louis/EmuELEC 100%[===================>]   3.85M  3.24MB/s    in 1.2s

2021-06-23 06:22:05 (3.24 MB/s) -
'/home/louis/EmuELEC/sources/duckstation/duckstation-8b234c06.zip' saved
[4037418/4037418]

    WARNING      Incorrect checksum calculated on downloaded file: got
0ab866981b13150c6e67336317eed002f26e5b26d2955c73d8eb3c6d67cc58fb wanted
3a986bc77837e92a116413956c6863b876effb018dbfd5da962d26ebd0f1c4b1
--2021-06-23 06:22:05--
http://sources.libreelec.tv/mirror/duckstation/duckstation-8b234c06.zip
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80...
connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location:
https://sources.libreelec.tv/mirror/duckstation/duckstation-8b234c06.zip
[following]
--2021-06-23 06:22:06--
https://sources.libreelec.tv/mirror/duckstation/duckstation-8b234c06.zip
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443...
[... 281 lines cut ...]
HTTP request sent, awaiting response... 404 Not Found
2021-06-23 06:22:55 ERROR 404: Not Found.

Cannot get duckstation sources :
https://www.duckstation.org/libretro/duckstation_libretro_linux_aarch64.zip
Try later!
FAILURE: scripts/install duckstation has failed!
[417/462] [FAIL] install duckstation

The following logs for this failure are available:
  stdout:
/home/louis/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.2/.threads/logs/420/stdout
  stderr:
/home/louis/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.2/.threads/logs/420/stderr

parallel: This job failed:
package_worker 3 420 462 'install duckstation'
Parallel build failure - see log for details. Time of failure: Wed Jun 23
06:22:56 UTC 2021
make: *** [Makefile:12: image] Error 1

Can you help?

Regards
Louis

earth2004, 2021-06-23:
Just found the package.mk of duckstation, problem fixed, thanks!