EmuELEC/EmuELEC #647: OGS vertical screen ratio 3:4 not available 3:2 and 21:9 have the same result

mbroers, 2021-06-23:
**Describe the bug**
OGS provides a 16:9 screen with a vertical screen rotate option. When vertical mode is enabled the full screen is used, which I believe results in a 9:16 ratio. The two menu options to affect the aspect ratio 3:2 and 21:9 have the same result, which pinches the screen horizontally and further distorts the vertical screen.

My understanding is that vertical games used a 4:3 monitor rotated to be 3:4. It would be awesome if 3:4 was an option and provided the proper aspect ratio while in vertical mode.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to advanced game options
2. Select ENABLE VERTICAL = ON
3. Select any of the VERTICAL ASPECT RATIO settings: 16:9, 3:2, 21:9 
4. No 3:4 "proper" aspect ratio achieved only full screen with distortion (16:9), and similar (same?) full vertical screen with compressed horizontal ratio (3:2 and 21:9)

**Expected behavior**
Im not sure if 3:2 is the proper way to label what I am expecting or if there should be an option for 3:4

**Screenshots**
[image]

[image]

[image]

**EmuELEC version:**
 - Version 4.1
 - Version 4.2
 
**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: Rockchip
 - Brand: Hardkernel
 - Model: Odroid Go Super

shantigilbert, 2021-06-25:
I am not sure I 100% understand the issue, you want to play a 4:3 game (or 3:4) in vertical mode?

qbertaddict, 2021-06-25:
i will test om a bit. Make sure if using the ratio's and rotation you turn off integer scale

mbroers, 2021-06-26:
[image]

Im probably just confusing things by trying to explain it so I just drew a picture. I would expect there to be a vertical aspect ratio mode 3:4 that looks like my diagram I think.  All existing options result in the full usage of the vertical length of the screen.

shantigilbert, 2021-06-26:
Yeah but I don't see the point in having vertical at 4:3 (3:4 in your case) at the moment this is not possible at least not from ES, it is only enabled to have the vertical resolutions, 4:3 is not a vertical resolution (is it?).  I guess if people want it, then I can add it.

shantigilbert, 2021-06-26:
Can you confirm that changing the ratio to 4:3 in Retroarch settings fixes this for you?

mbroers, 2021-06-27:
I must not be explaining it properly - Its totally possible I am conflating terms by calling the ogs screen 16:9. I'll try the retroarch setting.. 

Normal horizontal arcade games do not stretch to fill the whole OGS screen horizontally because the art would be distorted, so why would vertical games fill the whole screen vertically?  That is the point I am trying to make. 

All the vertical modes stretch to fill the whole screen but I would expect there to be one that looks like my drawing to maintain normal aspect ratio and not distort the art.

shantigilbert, 2021-06-27:
I understand what you are saying (I think), but that is not how it works, the image does not "scale" accordingly it will try to adjust to the ratio that it is set, if this means it will squish it or expand it then it will do it. This is because the ratio is forced and the image will fill that ratio. You can also see this in horizontal mode, If you force a 4:3 game to be in 16:9 it will fill the screen. You can set it to "Core Provided" or "4:3" if you wish to see it how you imagine.  

Either way, if you can confirm that setting it to "4:3" and/or "Core Provided" fixes it I have a commit ready.

mbroers, 2021-06-27:
[image]
this is 21x9 via the retroarch menu and i -think- it is what I was after.

mbroers, 2021-06-27:
[image]
And this is what 21x9 looks like from advanced game options ES menu

shantigilbert, 2021-06-27:
OK, so the issue is that it is not respecting the selection. I see, just to make it more clear, can you post a screenshot of your settings in ES for that game? is integer scale selected? or did you just select vertical mode on and the ratio? 

Also, can you post a screenshot of the video settings in Retroarch when you do this? I do not have my OGS handy so I cannot test, but I need to understand what is going on so I can fix it.

mbroers, 2021-06-28:
setting 21:9 in ES, no integer scale just vertical mode and vertical ratio:
[image]

result looks like this:
[image]

then go into retroarch menu video output settings, scaling, and set ratio to 21:9:
[image]

result looks like what i am after i believe:
[image]

shantigilbert, 2021-06-28:
OO, so the issue is that it is not respecting the ratio you set in ES. 

Can you show me one last photo, aspect ratio as it appears before you manually change it to 21:9 ?

mbroers, 2021-06-28:
When I set the game to ES Vertical and 21:9 in retroarch it initially has the Aspect Ratio as 4:3

shantigilbert, 2021-06-28:
Ok, thanks with this info I think I know what the issue is.

mbroers, 2021-06-28:
In the meantime where might I find the retroarch config files?  I might want to quickly manually set all my vertical games to 21:9..

shantigilbert, 2021-06-28:
> In the meantime where might I find the retroarch config files? I might want to quickly manually set all my vertical games to 21:9..

Not possible to change it manually, it will be overwritten by ES at launch. I am going to push a fix, but you will need to wait for an update.

commit 15bd9735 (2021-06-28): Fix vertical mode not respecting index ratio

Fixes #647 but further testing is needed
closed 2021-08-20 by shantigilbert