EmuELEC/EmuELEC #651: Supertux 2 on the Gameforce-CHI initial video settings not correct and do not save once changed

qbertaddict, 2021-06-27:
**Describe the bug**
Launch on Supertuc2 from ports on the gameforce and the screen and aspect ration are off
**To Reproduce**
Steps to reproduce the behavior:
Launch game

**Expected behavior**
Correct ratio and resolution at launch

**EmuELEC version:**
 - Version 4.2

Device-Gameforce chi

here is a working config.

(supertux-config
  (profile 1)
  (show_fps #f)
  (show_player_pos #f)
  (show_controller #f)
  (developer #f)
  (confirmation_dialog #f)
  (pause_on_focusloss #t)
  (custom_mouse_cursor #t)
  (integrations
    (hide_editor_levelnames #f)
  )
  (editor_autosave_frequency 5)
  (transitions_enabled #t)
  (locale "")
  (repository_url "")
  (video
    (fullscreen #t)
    (video "auto")
    (vsync #t)
    (fullscreen_width 640)
    (fullscreen_height 480)
    (fullscreen_refresh_rate 60)
    (window_width 640)
    (window_height 480)
    (window_resizable #t)
    (aspect_width 4)
    (aspect_height 3)
    (magnification 0)
  )
  (audio
    (sound_enabled #t)
    (music_enabled #t)
    (sound_volume 100)
    (music_volume 50)
  )
  (control
    (keymap
      (sdl2 #t)
      (jump-with-up #f)
      (map
        (key 8)
        (control "remove")
      )
      (map
        (key 13)
        (control "menu-select")
      )
      (map
        (key 27)
        (control "escape")
      )
      (map
        (key 32)
        (control "jump")
      )
      (map
        (key 94)
        (control "console")
      )
      (map
        (key 112)
        (control "start")
      )
      (map
        (key 127)
        (control "peek-left")
      )
      (map
        (key 1073741882)
        (control "cheat-menu")
      )
      (map
        (key 1073741883)
        (control "debug-menu")
      )
      (map
        (key 1073741896)
        (control "start")
      )
      (map
        (key 1073741898)
        (control "peek-up")
      )
      (map
        (key 1073741901)
        (control "peek-down")
      )
      (map
        (key 1073741902)
        (control "peek-right")
      )
      (map
        (key 1073741903)
        (control "right")
      )
      (map
        (key 1073741904)
        (control "left")
      )
      (map
        (key 1073741905)
        (control "down")
      )
      (map
        (key 1073741906)
        (control "up")
      )
      (map
        (key 1073741912)
        (control "menu-select")
      )
      (map
        (key 1073742048)
        (control "action")
      )
      (map
        (key 1073742050)
        (control "action")
      )
    )
    (joystick
      (dead-zone 8000)
      (jump-with-up #f)
      (use-game-controller #t)
      (map
        (button 0)
        (control "menu-select")
      )
      (map
        (button 1)
        (control "action")
      )
      (map
        (button 4)
        (control "peek-left")
      )
      (map
        (button 5)
        (control "peek-right")
      )
      (map
        (button 6)
        (control "start")
      )
      (map
        (axis -2)
        (control "up")
      )
      (map
        (axis -1)
        (control "left")
      )
      (map
        (axis 1)
        (control "right")
      )
      (map
        (axis 2)
        (control "down")
      )
    )
  )
  (addons
  )
)

It does not save when you exit the game and sets everything back to old settings when you launch. setting to read only does not work

shantigilbert, 2021-06-27:
> t does not save when you exit the game and sets everything back to old settings when you launch

This is the only thing that worries me, the resolution was a known problem and we agreed to just let the user set this to whatever they like. 

But when you say its not saving, what do you mean? the settings save fine for me when I change them in-game, are you not doing it in-game?

qbertaddict, 2021-06-27:
I set them in game and even save a new profile. as soon as i launch it again it goes back to the old settings. The file i posted are what the config file was before i exit normally from the game and are good for the gameforce-chi

shantigilbert, 2021-06-27:
I cannot reproduce the not saving issue... can you walk me trough the steps like you do it? I change resolution, quit the game and the settings are saved as expected.

shantigilbert, 2021-06-27:
Nevermind, I see what is the issue

commit e4e76224 (2021-06-27): SuperTux2: Fix sed in launch script

Fixes #651
closed 2021-08-20 by shantigilbert