EmuELEC/EmuELEC #652: Emulator Advmame from Emulation Station showing Black/Blank Screen.

Langerz82, 2021-06-27:
**Describe the bug**
When launching MAME games into the AdvMame emulator a blank screen shows, but the ROMs audio and input are fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to MAME ROM Folder.
2. In ES Settings, select the Advmame emulator.
3. Select any MAME game and run.

**Expected behavior**
The video too display.

**Screenshots**
None

**EmuELEC version:**
v4.2

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Beelink GT King Pro

**Additional context**
ES Video Setting:
720p60

Running through a hdmi output.

Here are my relevant advmame.rc settings:
```
device_alsa_device sdl
device_color_bgr15 no
device_color_bgr8 no
record_sound no
record_video no
sound_latency 0.10
sound_normalize no
display_rgbeffect scan2horz
display_vsync no
device_sound sdl
device_video auto
device_video_clock 10-180 / 30-80 / 50-70
```
device_video sdl <-- does not work either.
I think it could be the device_video_clock though I'm not sure what it should be for HDMI output.

shantigilbert, 2021-06-27:
Thanks I can confirm this is happening, and I have fixed it in DEV. 

Issue is that DISPLAY is being set before launch and it should be empty.

Langerz82, 2021-06-28:
Ace, thanks man, I look forward to using Advmame next version.

shantigilbert, 2021-06-28:
You can play it on this version if you download the file https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emulators/advancemame/bin/advmame.sh and copy it to `/emuelec/bin` (or `ConfigFiles/emuelec/bin` if using network shares)

Langerz82, 2021-06-28:
I put it in ".config/emuelec/bin/" but launching a mame game screen is still blank and checking the logs it seems "/usr/bin/advmame.sh" is being called.

shantigilbert, 2021-06-28:
Well I guess you need to wait for a new version, sorry about that.

Langerz82, 2021-06-29:
Can you access the linux filesystem through ssh?

Langerz82, 2021-07-02:
I unsquashed the SYSTEM file, and could modify the filesystem that way then re-squashed it, the mount seems alright, how is the system md5 calculated? Just the original SYTEM files md5?

Langerz82, 2021-07-02:
Hey I think:
```
            if [ -f /ee_s905 ]; then
                echo "device_video_modeline 1280x720-60 91.517 1280 1440 1531 1691 720 810 812 902 +hsync +vsync" >> $CONFIG_DIR/advmame.rc
            fi
```
Should just be:
```
                echo "device_video_modeline 1280x720-60 91.517 1280 1440 1531 1691 720 810 812 902 +hsync +vsync" >> $CONFIG_DIR/advmame.rc
```

I haven't tested it yet, it's late and I will tomorrow, because my chip is S922X the condition might not work.

Langerz82, 2021-07-02:
Yep , I removed the conditional and now the video is not zoomed and displaying properly, cheers all.

Langerz82, 2021-07-03:
The game clock was running too fast in my advmame.rc config file because vsync was set to true.
```
display_vsync yes
```
should be:
```
display_vsync no
```
The ElecEmu default advmame.rc file should default to "no" so you should only need to worry about it if you change it or using a modified advmame.rc file. 

Cheers.

closed 2021-08-20 by shantigilbert

Langerz82, 2021-10-14:
Hey, an easier way would be to just add the line manually into advmame.rc by ssh'ing in and running:
```
       echo "device_video_modeline 1280x720-60 91.517 1280 1440 1531 1691 720 810 812 902 +hsync +vsync" >> /storage/.advance/advmame.rc
```
It should enter the video mode for 720p.

Lilearnscode, 2021-10-15:
I know this is a very idiot problem in a tech forum like this. but thank
you for spending time on my post.  I dont know how to find and edit the
file through SSH (I tried to google some videos and did find how to use SSH
connecting to it)...

commit bc2bc473 (2021-06-27): advmame.sh: unset DISPLAY for Amlogic and Amlogic-ng

Fixes #652