EmuELEC/EmuELEC #653: Feature - Some Additional Retroarch Cores request.

Langerz82, 2021-06-28:
Can we also get the following Cores for RetroArch:
Parallel64 - https://github.com/libretro/parallel-n64
(Capcom 1) - cps1 - https://github.com/libretro/fbalpha2012_cps1
(Capcom 2) - cps2 - https://github.com/libretro/fbalpha2012_cps2
Apologies if cps1 and cps2 is added, but I could not switch to emulator in EmulationStation under advances settings for there respective folders.

Langerz82, 2021-06-28:
The NeoGeo emulator seems to be missing too. FBNeo can only run some games.

AlienatedDiego, 2021-06-28:
Strange, fbneo is up to date and runs my arcade and neogeo 0.2.97.44 games correctly. Could it be that you are missing neogeo.zip in roms / neogeo, or that you are using a mixed / old romset?

shantigilbert, 2021-06-28:
Why do we need these? 
Parallel is already included, and fbneo seems to emulate all the games for CPS1 and CPS2

Langerz82, 2021-06-29:
Parallel was generating an error, let me check it again.
CPS1 and CPS2, I think my ROM set was bad let me check them properly.

Langerz82, 2021-06-29:
My CPS roms have invalid CRC's I'm trying to fix them.

Regarding Parallel64, I get this error when switching it in EmulationStation, under Advanced Settings >> Emulator:
```
EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.2 (Amlogic-ng.aarch64)
PLATFORM: n64
ROM NAME: /storage/roms/n64/Bug's Life, A (USA).zip
BASE ROM NAME: Bug's Life, A (USA).zip
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/n64/Bug's Life, A (USA).zip
2nd Argument: -Pn64
3rd Argument: --core=parallel_n64_32b
4th Argument: --emulator=libretro
Full arguments: /storage/roms/n64/Bug's Life, A (USA).zip -Pn64 --core=parallel_n64_32b --emulator=libretro --controllers=-p1index 0 -p1guid 050000004c0500006802000000000000 
**Run Command is:
retroarch32 -v -L /tmp/cores/parallel_n64_32b_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/n64/Bug's Life, A (USA).zip**
Emulator Output is:
**[WARN] --libretro argument "/tmp/cores/parallel_n64_32b_libretro.so" is neither a file nor directory. Ignoring.**
[INFO] RetroArch 1.9.3 (Git 122be0cc64)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON ASIMD
[INFO] Built: May 24 2021
[INFO] Version: 1.9.3
[INFO] Git: 122be0cc64
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] [Core]: Frontend is built for dynamic libretro cores, but path is not set. Cannot continue.
[ERROR] Fatal error received in: "init_libretro_symbols()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core symbols..
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
exit 1
```

shantigilbert, 2021-06-29:
you need to rename the file `/tmp/cores/parallel_n64_32_libretro.so` to `/tmp/cores/parallel_n64_32b_libretro.so`

I am going to close this as it is not really an issue/bug, and we determined that those cores are not needed. If you need more help, you can open a thread on the forum or discuss it in Discord.

closed 2021-06-29 by shantigilbert

Langerz82, 2021-06-29:
Ah thanks for the solution for a parallel fix.

Hey, might just add. With cps1 I had more success with roms using MAME 2003 Plus.
With cps2 fb-alpha2012 works very well for most if not all roms.

Langerz82, 2021-06-29:
Oh I renamed the parallel file, I didnt know we could access the linux filesystem through ssh.