EmuELEC/EmuELEC #654: Emulator - BlueMsx on MSX roms - ElecEmu Error.

Langerz82, 2021-06-29:
**Describe the bug**
Running a MSX Rom in BlueMSX generates the following error:
```
1st Argument: /storage/roms/msx/10-Yard Fight (Japan).zip
2nd Argument: -Pmsx
3rd Argument: --core=bluemsx
4th Argument: --emulator=libretro
Full arguments: /storage/roms/msx/10-Yard Fight (Japan).zip -Pmsx --core=bluemsx --emulator=libretro --controllers=-p1index 0 -p1guid 050000004c0500006802000000000000 
Run Command is:
retroarch -v -L /tmp/cores/bluemsx_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/msx/10-Yard Fight (Japan).zip
Emulator Output is:
[INFO] RetroArch 1.9.4 (Git c226bd87f4)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Jun 13 2021
[INFO] Version: 1.9.4
[INFO] Git: c226bd87f4
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/bluemsx_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/blueMSX/blueMSX.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/blueMSX/msx.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/blueMSX/10-Yard Fight (Japan).cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: Remap directory: "/storage/.config/retroarch/config/remaps".
[INFO] [Overrides]: Redirecting save file to "/storage/.config/retroarch/saves/10-Yard Fight (Japan).srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/savestates/msx/10-Yard Fight (Japan).state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [CONTENT LOAD]: Core requires uncompressed content - extracting archive to temporary directory.
[INFO] [CONTENT LOAD]: Content successfully extracted to: /storage/roms/msx/10-Yard Fight (Japan).rom
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/.config/retroarch/system".
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [CONTENT LOAD]: Removing temporary content file: /storage/roms/msx/10-Yard Fight (Japan).rom.
[ERROR] [CONTENT LOAD]: Failed to load content
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Core Options]: Saved core options file to "/storage/.config/retroarch/config/blueMSX/blueMSX.opt"
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
The framebuffer device was opened successfully.
vinfo.yoffset=0
exit 1
bluemsx libretro /storage/roms/msx/10-Yard Fight (Japan).zip
MSX missing BIOS - Could not find all BIOS: 
please make sure you copied the files into the corresponding folder 
> MSX
MISSING - bios/Machines/MSX2 - Arabic/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - Russian/hardwareconfig.xml
MISSING - bios/Machines/Turbo-R/hardwareconfig.xml
MISSING - bios/Machines/MSX - Brazilian/hardwareconfig.xml
MISSING - bios/Machines/MSX - Korean/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - French/hardwareconfig.xml
MISSING - bios/Machines/MSX - German/hardwareconfig.xml
MISSING - bios/Machines/MSX/hardwareconfig.xml
MISSING - bios/Machines/MSX - Swedish/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - Japanese/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - Only PSG/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - Korean/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - Brazilian/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - Spanish/hardwareconfig.xml
MISSING - bios/Machines/MSX2+/hardwareconfig.xml
MISSING - bios/Machines/MSX2 - German/hardwareconfig.xml
MISSING - bios/Machines/MSX2/hardwareconfig.xml
MISSING - bios/Machines/MSX - Japanese/hardwareconfig.xml
exit bios 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to MSX Game List
2. Select any MSX Game and run under libretro blueMSX emulator.
3. Error should occur.

**Expected behavior**
The games to run ok.

**EmuELEC version:**
4.2

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Beelink GT King Pro

**Additional context**
I've noticed that the bluemsx "storage/bios/Machines/Shared Roms" is possibly not being correctly parsed because I had to add those files to the "storage/bios/Machines/MSX", "storage/bios/Machines/MSX2" and "storage/bios/Machines/MSX2+" folders for the roms to be detected.

I compared my blueMSX bios with a gamebox with previous version EmuElec 3.8 (I think) and no "hardwareconfig.xml" is present in it's folders and the gamebox loads the games fine.

shantigilbert, 2021-06-29:
Please use the forum or discord for these type of issues, Github should be used for bugs, real bugs no user error.

closed 2021-06-29 by shantigilbert

Langerz82, 2021-06-30:
Hello,
I have the BlueMSX BIOS copied in the BIOS folder, I used the same version that was in an working ElecEmu and it's throwing up these errors when it should work like the previous versions.
I tried to find the hardwareconfig.xml files for the folders but couldn't find any online. Have you got the links to the correct BIOS with the hardwareconfig.xml's?

Langerz82, 2021-06-30:
Solved see:
https://emuelec.discourse.group/t/emulelec-4-2-bluemsx-bios-missing-hardwareconfig-xml-in-several-machine-folders-solved/186/4?u=langerz82