EmuELEC/EmuELEC #657: [EmuElec 4.2][Amlogic-ng] - System Volume Always Resets to 98% on Reboot

zerojay, 2021-07-01:
The system volume set by the user through EmuElec's UI is not remembered across reboots and resets to 98% each time.

Repro Steps:
User presses Start to bring up EmuElec menu.
User selects Sound Options.
User adjusts System Volume to any volume other than 98%.
User selects Reboot System from EmuElec menu.
After rebooting, open the EmuElec menu and go to Sound Options.
User sees and hears that system volume is set to 98% unexpectedly.

Expected Behavior:
Upon rebooting, EmuElec should restore the last selected system volume that the user selected. 

EmuELEC version: 4.2
Device: Odroid N2 - Amlogic-ng - 4GB RAM version

shantigilbert, 2021-07-01:
It never occurred to me that this could also affect other devices, it is already fixed handhelds, I will check it for Amlogic.

Thanks for reporting

AlienatedDiego, 2021-07-01:
Tested on Amlogic ng (s905x3) the overall system volume always sets 100% on each reboot, it does not seem to affect the music volume that if it keeps the user value after each reboot

shantigilbert, 2021-07-02:
@zerojay Changing the volume works for you, I mean do you notice the volume lowering? AFAIK the volume slider on ES has never worked on Amlogic, so I am curious if it works for you.

zerojay, 2021-07-02:
Some additional details for you: 

I'm using my Odroid N2 in an arcade machine. This machine has a monitor which has no speakers, so I cannot use HDMI audio out. Instead, I am using the audio out jack on the board for all my audio. To do so, I must set my audio device to 0,1 in EmuElec. When using analog audio through the audio jack with audio device set to 0,1 - yes, indeed, the system volume bar does function for setting the volume from emulationstation just fine when I'm using analog audio. (I cannot speak for HDMI audio at all, however.)

I came to EmuElec from Batocera, where I very quickly discovered that wifi through an external dongle (wireless access points are visible but cannot be connected to) and analog audio out (even with audio device set to 0,1) were both broken on the N2.

shantigilbert, 2021-07-02:
I have looked a bit into this, and I think I know why the issue happens, but I do not know of a way to fix it unfortunately. 

The issue seems to be that on boot the volume gets reset to max by the system, and since ES reads the volume value from what is it in the system it resets itself as well. 

I will try and look for a solution.

shantigilbert, 2021-07-02:
This will be fixed on the next release, thanks for the report

zerojay, 2021-07-02:
Confirmed fixed with the script you passed to me, thank you.

commit 8bf0fd01 (2021-07-02): Fix volume resets to 98% using AV

Fixes  #657
closed 2021-08-20 by shantigilbert