EmuELEC/EmuELEC #659: [EmuElec 4.2][Odroid Go Super] Blank sceen on second boot (and onward)

pixl-creator, 2021-07-02:
**Describe the bug**
After the first shutdown and onward, Emuelec boots to a blank screen.
It's possible to get the screen back by starting an emulator then quitting by going in blind.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a fresh EmuElec install 
2. Power down the Odroid Go Super through the system menu
3. Power it back on and wait for Emuelec to boot
4. See error

**Expected behavior**
The main menu is expected to appear after boot

**Screenshots**
N/A

**EmuELEC version:**
 - Version 4.2

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: Rockchip
 - Brand: Odroid
 - Model Odroid Go Super
 - Version 1

**Additional context**
I've read online that it's because the screen resolution isn't set correctly on boot but don't quote me on that.
Likely a screen resolution issue though.

british-choi, 2021-07-02:
what is error?

pixl-creator, 2021-07-02:
Nothing on screen. Menu is still responsive.

british-choi, 2021-07-03:
> […](#)

hmm ..
I cannot reproduce
change your sd card, and try again

pixl-creator, 2021-07-03:
Flashed the sd card a second time and it's still happening.

shantigilbert, 2021-07-03:
> The sd card has no issues.

Have you tried another SD card? 

I cannot reproduce this at all, Odroid Go Advance, Odroid Go Super and GameForce all work as they should

pixl-creator, 2021-07-03:
No, I haven't.
The sd card has no issues and works perfectly fine.
It's possible to get the screen back up by launching an emulator by
navigating the menu while the screen is blank.

shantigilbert, 2021-07-03:
> No, I haven't.
> The sd card has no issues and works perfectly fine.

Please use another SD card and test, don't assume. There is no issue that is reproducible here, I am sure lots of other people would have reported it already.

shantigilbert, 2021-07-03:
If you want to make sure there is not a HW issue on your device you might want to try Batocera, or another distro that runs on the OGS.

pixl-creator, 2021-07-03:
I used Lakka before (stable and nightly) and had zero issues with it with
the same hardware and sd card.

shantigilbert, 2021-07-03:
Then I have no idea, sorry, I cannot reproduce it here.

pixl-creator, 2021-07-03:
I've included a log there : http://ix.io/3rSg
The screen went back up when I launched a game.

shantigilbert, 2021-07-03:
Sorry, the log will not help me identify this issue, and you seem to refuse my advice to test on another SD card. So I cannot help you.

pixl-creator, 2021-07-05:
Found a temporary fix by booting to xmb first.

closed 2021-07-06 by shantigilbert