EmuELEC/EmuELEC #660: PPSSPP build failed

keeno1982, 2021-07-02:
Please help to see

-- Up-to-date: /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/ppsspp-79e3d1c5f9c47899a786d8e3        053df8557e006b96/.aarch64-libreelec-linux-gnueabi/assets/flash0/font/ltn10.pgf
-- Configuring incomplete, errors occurred!
See also "/home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/ppsspp-79e3d1c5f9c47899a786d8e3053df        8557e006b96/.aarch64-libreelec-linux-gnueabi/CMakeFiles/CMakeOutput.log".
See also "/home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/ppsspp-79e3d1c5f9c47899a786d8e3053df        8557e006b96/.aarch64-libreelec-linux-gnueabi/CMakeFiles/CMakeError.log".
FAILURE: scripts/install ppsspp has failed!
[324/408] [FAIL] install ppsspp

The following logs for this failure are available:
  stdout: /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/.threads/logs/323/stdout
  stderr: /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/.threads/logs/323/stderr

parallel: This job failed:
package_worker 3 323 408 'install ppsspp'
Parallel build failure - see log for details. Time of failure: Fri Jul  2 10:10:31 CST 2021
make: *** [Makefile:12: image] Error 1

CMakeError.log below:

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/ppsspp-79e3d1c5f9c47899a786d8e3053df8557e006b96/.aarch64-libreelec-linux-gnueabi/CMakeFiles/CMakeTmp

Run Build Command(s):/home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/toolchain/bin/ninja cmTC_b7855 
[1/2] Building C object CMakeFiles/cmTC_b7855.dir/CheckSymbolExists.c.o
[2/2] Linking C executable cmTC_b7855
FAILED: cmTC_b7855 
: && /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/toolchain/bin/aarch64-libreelec-linux-gnueabi-gcc -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -march=armv8-a+crc+crypto+fp+simd -mtune=cortex-a53 -Wl,--as-needed -fuse-ld=gold CMakeFiles/cmTC_b7855.dir/CheckSymbolExists.c.o  -o cmTC_b7855   && :
CMakeFiles/cmTC_b7855.dir/CheckSymbolExists.c.o:CheckSymbolExists.c:function main: error: undefined reference to 'pthread_create'
CMakeFiles/cmTC_b7855.dir/CheckSymbolExists.c.o:CheckSymbolExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

File /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/ppsspp-79e3d1c5f9c47899a786d8e3053df8557e006b96/.aarch64-libreelec-linux-gnueabi/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/ppsspp-79e3d1c5f9c47899a786d8e3053df8557e006b96/.aarch64-libreelec-linux-gnueabi/CMakeFiles/CMakeTmp

Run Build Command(s):/home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/toolchain/bin/ninja cmTC_4e99d 
[1/2] Building C object CMakeFiles/cmTC_4e99d.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTC_4e99d
FAILED: cmTC_4e99d 
: && /home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/toolchain/bin/aarch64-libreelec-linux-gnueabi-gcc -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a53 -mcpu=cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3 -DCHECK_FUNCTION_EXISTS=pthread_create  -march=armv8-a+crc+crypto+fp+simd -mtune=cortex-a53 -Wl,--as-needed -fuse-ld=gold CMakeFiles/cmTC_4e99d.dir/CheckFunctionExists.c.o  -o cmTC_4e99d  -lpthreads && :
/home/dietpi/emu/build.EmuELEC-Amlogic.aarch64-4.2/toolchain/lib/gcc/aarch64-libreelec-linux-gnueabi/8.3.0/../../../../aarch64-libreelec-linux-gnueabi/bin/ld.gold: error: cannot find -lpthreads
CMakeFiles/cmTC_4e99d.dir/CheckFunctionExists.c.o:CheckFunctionExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

shantigilbert, 2021-07-02:
How are you compiling? what OS?

closed 2021-07-07 by shantigilbert

keeno1982, 2021-07-08:
> 
> 
> How are you compiling? what OS?

Sorry for reply late, OS is Dietpi run on VMware.

keeno1982, 2021-07-08:
I'm trying rebuild V4.1 now, I remember that successed before. But V4.2 failed and failed.

shantigilbert, 2021-07-08:
> OS is Dietpi

I have never used that OS to compile, so I would not know if this is the real issue, but the whole system compiles fine under Ubuntu 18.04 which is what is recommended to used

shantigilbert, 2021-07-08:
Try adding: `export LDFLAGS="${LDFLAGS} -lpthread"` to the package.mk

keeno1982, 2021-07-09:
Wow~build successed!