EmuELEC/EmuELEC #664: 4.2 forced  update from 4.1 doesn't  work

sdgaey5qh5qh515h5, 2021-07-03:
**Describe the bug**

Forced update to 4.2 doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1) download the 4.2 release
2) copy it to /storage/.update using scp or rsync
3) go to menu and choose forced update
4) emulationstation will save metadata
5) screen will go black for a second
6) emulationstation will load again
7) update hasn't happened but the tar in .update was deleted

**Expected behavior**
update to 4.2

**Screenshots**
none

**EmuELEC version:**
4.1

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Super Console X / Amlogic S905X rev e

**Additional context**
Thank you.

shantigilbert, 2021-07-03:
> **Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
> Super Console X / Amlogic S905X rev e

Question, did you even read the link provided? 

I will link it again directly where it shows that device is not supported: https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices#special-note

shantigilbert, 2021-07-03:
I will still give you a hint, it will not make a difference for your since your device will still not work properly, but most likely your device is lacking the required space for update. It needs at LEAST 1 GB free after you copy the .tar file, because it needs space to expand the files and copy them. In the future, please ask for support on the manufacturer website for your device, I do not support those companies nor I will ever.

closed 2021-07-03 by shantigilbert