EmuELEC/EmuELEC #666: [EmuELEC 4.2][GameForce] Save state slot bumped when using "Save Snapshot"

ffreling, 2021-07-05:
**Describe the bug**
In the EmuELEC UI, if I load a game through the "Save Snapshot" feature (X button), then the save slot number is increased.
If I save without decreasing the slot number, it will create a new save state.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any game in your library
2. Launch it without snapshot (A button)
3. Check the save slot in RetroArch, it is 0
3. Save your state in default save slot 0
4. Quit game
5. Launch your game with the "Save Snapshot" feature (X button) and choose your save slot
6. Check the save slot in RetroArch, it is 1 (equal to the number of save states, can be more over time)

**Expected behavior**
The default save slot should match the one you choose when launching via "Save Snapshot".
It should not increase, otherwise by default you will increase your number of save states over time.

**EmuELEC version:**
 - Version 4.2

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - GameForce

shantigilbert, 2021-07-05:
This is intended. The idea is to not overwrite the save, since using auto load will also enable auto save.

ffreling, 2021-07-06:
I had "Auto-save" disabled in the settings, so I thought it was not active.
I understand now that going through "Snapshot" enables "Autosave".

Case closed :)

closed 2021-07-06 by ffreling