EmuELEC/EmuELEC #667: [EmuELEC 4.2][GameForce] D-pad issue after waking up from sleep

ffreling, 2021-07-06:
**Describe the bug**
After waking up from sleep, either D-Pad “left” is not working or it is always on.
Rebooting fixes the issue. No problem with the sticks though.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sleep
2. Wake up device
3. Press "left" on D-pad
4. Controls are not properly working

**Expected behavior**
Controls should work fine after waking up from sleep.

**EmuELEC version:**
 - Version 4.2 (fresh install)
 - Version 4.1 (fresh install)

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - GameForce Chi

**Additional context**
There is a thread in the forum about the issue: https://emuelec.discourse.group/t/my-gameforce-has-a-problem-pressing-dpad-after-resume/112/16
I created this issue to help tracking. Is there any log I could copy to help understanding the problem?

aqualime-bit, 2021-07-06:
Strange enough I have the same problem when exiting a game or sleeping. It's only with a Chinese basic SNES gamepad. My nicer ps3 style pads and others do not have the same issue.

shantigilbert, 2021-07-06:
Sorry pressed the wrong button

shantigilbert, 2021-07-18:
@ffreling How long do you keep your device in sleep? and is it in game? or just ES?

ffreling, 2021-07-18:
I kept it in sleep for a few seconds when testing, but also tried letting it sleep for minutes.
The issue is both in game and in ES.

shantigilbert, 2021-07-18:
Thanks, I am still looking into this, I cannot reproduce but I will keep trying.

1CuriousCrab, 2021-11-11:
I have somewhat the same issue, my D-pad sometimes think I'm pressing left when I quickly tap up or down. Any progress regarding this issue? I'm using Emuelec v4.3

den13501, 2022-02-28:
> I have somewhat the same issue, my D-pad sometimes think I'm pressing left when I quickly tap up or down. Any progress regarding this issue? I'm using Emuelec v4.3

yeah, I have same issue with you.

closed 2022-06-30 by ffreling