EmuELEC/EmuELEC #668: Running backup util gives tar error

bsturk, 2021-07-13:
**Describe the bug**
Running the recommended backup command

emuelec-utils ee_backup backup no

gives this error and quits:

tar: /storage/.emulationstation/scripts/drastic/config/*: No such file or directory
tar: /storage/.emulationstation/scripts/drastic/*.dsv: No such file or directory
tar: error exit delayed from previous errors

**To Reproduce**
Steps to reproduce the behavior:

emuelec-utils ee_backup backup no

**Expected behavior**
A backup tarball

**EmuELEC version:**
 4.0

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE]
Odroid Go Advance

**Additional context**
Attempting to update to 4.2... I can always just run tar on the .config dir and scp it off, but I want to ensure I get other useful config

shantigilbert, 2021-07-13:
> gives this error and quits

Those are very descriptive errors...Those folders do not exist, I take it you do not have drastic installed, nothing to worry about

The tarball should be in `/storage/roms/backup` or `/var/media/EEROMS/backup` if you have an external HDD 

Check those paths and please close this issue if its there.

bsturk, 2021-07-13:
Hi, yes the tarball is there, tar seems to just treat any missed file/folder specified as an error, but keeps on going.

closing...

closed 2021-07-13 by bsturk