EmuELEC/EmuELEC #669: [Emuelec 4.2] [S905X3 rev C] Can't pair a PS3 controller

jbonina, 2021-07-16:
Hi!
I have a TX3 4 Gb S905X3 rev C box. I installed Emuelec 4.2 and I can't pair it with a PS2 controller.
Previously, I had Emuelec 3.4 and the controller was paired and working.

I followed instructions at https://github.com/EmuELEC/EmuELEC/wiki/Using-PS3-Gamepads  but no controller appears as detected or paired in the bluetooth controllers list in the Emuelec UI.

If I run the bluetoothctl commands on the cmd line, it will trust the controller and sometimes it will connect to the controller, sometimes it won't. In these cases, the controller will appear in the bluetooth controllers list in the Emuelec UI but I won't be able to use it.

shantigilbert, 2021-07-18:
Is this an original Ps3 gamepad?

jbonina, 2021-07-23:
This can be closed. It was a bad USB cable.

closed 2021-07-23 by shantigilbert