EmuELEC/EmuELEC #672: Several emulators don't work

6david99, 2021-07-18:
ResidualVM - don't work
ScummVM - emulator Scummvmsa no controller response
SPECTRUM - no virtual keyboard
Amstrad GX4000 - no show in emulationstation
File manager - in emulationstation menu press F1 don't work
Wii - no controller config

shantigilbert, 2021-07-18:
Can you please follow the issue templates?

6david99, 2021-07-18:
anyone who downloads the latest version 4.2 and tries it will see that it does not work, it is not my problem, it is a general problem

shantigilbert, 2021-07-18:
Never said it was your problem I said you need to fill in the template... I have no idea what gamepad, what device, or anything else. 

The only one that is "general" is this, "Wii - no controller config" as it is a known issue. the others I will ignore until you provide logs, or fill out the template.  no logs, no problem.

6david99, 2021-07-18:
I'm sorry I don't know how to fill in the ticket, I have already mentioned the problems, if anyone is curious you can see it, it is not with any gamepad, the errors are with the keyboard.

shantigilbert, 2021-07-18:
OK, well when you feel like helping out so I can help you, feel free to reopen the ticket.

closed 2021-07-18 by shantigilbert