EmuELEC/EmuELEC #680: setres.sh unwriteable

ducdreamer, 2021-07-27:
Hello there,
I have made a new install of emuelec 4.2
And now I try to edit the setres.sh in usr/bin
to resize the screen size to my tv
But now i cannot write the modified file to de sd card
I have tries to modify the file attributes to 777 bit the system wil not let me write the modified file back
In my old install the file was found in /storage/.config/emuelec/scripts/setres.sh
now it is etc/bin/setres.sh

closed 2021-07-27 by shantigilbert

ducdreamer, 2021-07-27:
I am sorry but where is this template i must follow?

shantigilbert, 2021-07-27:
> I am sorry but where is this template i must follow?
When you open an issue you get several options, you went to the trouble deleting all the text without reading it? 

Anyways, for your issue: copy setres.sh to /emuelec/bin and edit it there.