EmuELEC/EmuELEC #690: Audio stuttering in lzdoom

miwasp, 2021-08-16:
**Describe the bug**
Audio in lzdoom is more or less stuttering or choppy.

**To Reproduce**
Steps to reproduce the behavior:
1. Copy doom.wad and brutalv21.pk3 to /storage/roms/ports/doom-mods (https://www.moddb.com/mods/brutal-doom)
2. Create a file brutal.doom in /storage/roms/ports/doom with contents

> GAMETYPE=doom
> SUBDIR=
> PARAMS=-iwad /storage/roms/ports/doom-mods/doom.wad -file /storage/roms/ports/doom-mods/brutalv21.pk3

3. Restart ES
4. Start brutal out of section Doom
5. First you can hear the music without problems, but in game you can't even hear your gunfire

**Expected behavior**
Audio should not stutter, gunfire should be hearable.

**EmuELEC version:**
4.2 and 4.3 TEST

**Device (please complete the following information):**
 - OGA
 - OGS
 - Gameforce
 - According to Discord also Amlogic devices

**Additional context**
I already tested:
- updated lzdoom to 3.88a, same problem
- updated mpg123 to 1.28.2, same problem
- tested EVERY audio setting in lzdoom, always same problem
- tested the same wads and pk3s with RetroOz 0.50 -> perfect sound

**Error messages**
If i start lzdoom directly from the console
1. systemctl stop emustation
2. /usr/bin/lzdoom -iwad /storage/roms/ports/doom-mods/doom.wad -file /storage/roms/ports/doom-mods/brutalv21.pk3

i can see many error messages like these:

> Note: Illegal Audio-MPEG-Header 0xffcfffcf at offset 18469.
> Note: Trying to resync...
> Note: Skipped 221 bytes in input.
> [/home/miw/EmuELEC/build.EmuELEC-OdroidGoAdvance.aarch64-4.2/mpg123-compat-1.25.10/src/libmpg123/layer1.c:30] error: Illegal bit allocation value.
> [/home/miw/EmuELEC/build.EmuELEC-OdroidGoAdvance.aarch64-4.2/mpg123-compat-1.25.10/src/libmpg123/layer1.c:174] error: Aborting layer I decoding after step one.

or these:

> Note: Illegal Audio-MPEG-Header 0xd1004e02 at offset 11200.
> Note: Trying to resync...
> Note: Skipped 310 bytes in input.
> 
> Warning: Big change from first (MPEG version, layer, rate). Frankenstein stream?

Any ideas on this?

shantigilbert, 2021-08-17:
> Any ideas on this?

I don't but it looks like it has something to do with libmpg123.

miwasp, 2021-08-17:
It happens on all kinds of mods. Brutal Doom was just an example to easily reproduce it.

shantigilbert, 2021-08-17:
Can you test the binary from where you said works on EmuEELEC?

miwasp, 2021-08-17:
Do ArkOS binaries run in EmuELEC?

shantigilbert, 2021-08-17:
I have no idea, that is why I asked.

miwasp, 2021-08-17:
Ok, i will test it after work…

miwasp, 2021-08-17:
After fiddling with a few libs i was really able to run the working ArkOS binary in EmuELEC. Unfortunately audio is still choppy, so its not a problem with lzdoom itself.

shantigilbert, 2021-08-17:
Well that is a start, but I still do not know how to debug this.

commit 74362ac9 (2021-08-19): Bump several audio packages

This also fixes #690
closed 2021-08-19 by shantigilbert