EmuELEC/EmuELEC #692: EmuELEC 4.2 cannot connect to a WiFi network if the password has a dollar sign ($) in it

Dr-KerbalMD, 2021-08-18:
**Describe the bug**
EmuELEC 4.2 cannot connect to a WiFi network if the password has a dollar sign ($) in it.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a wireless network with a WPA2-Personal [AES] password that has a $ in the password
2. On the device running EmuELEC, go to Network Setting, enable WiFi, select the SSID of the network, and enter the password.
3. The IP address will populate with a long alphanumeric string separated by several colons. The device will not connect to the internet.
4. Check your router logs. On my Netgear Nighhawk, I got: 

> WLAN access rejected: incorrect security] from MAC address **:**:**:**:**:**, Tuesday, August 17, 2021 22:48:07

**Expected behavior**
The device should connect to the internet.

**EmuELEC version:**
 - Version 4.2

**Device (please complete the following information):**
 - SOC: RK3326
 - Brand: Hardkernel
 - Model: ODROID-GO Super
 - Version: Not sure. It's the updated model with the larger joysticks.
 
**Additional Context**

My router has a guest network with a different password and EmuELEC connects to that network without issue. I even tried removing the $ from the password of the primary network, and EmuELEC connects without issue. But as soon as I add the $ back in the password,  it fails and the router rejects the authentication attempt from EmuELEC.

shantigilbert, 2021-08-18:
What if you put your password on `emuelec.conf` ? https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/config/emuelec/configs/emuelec.conf#L63

Or by using ee_defaults.txt as explained on this commit: https://github.com/EmuELEC/EmuELEC/commit/9358efe01af501326526ce479ca6c5f55932e34a

Dr-KerbalMD, 2021-08-18:
Thanks for replying so quickly. Where do I find emuelec.conf? I cannot find it on either EMUELEC or EEROMS when I mount the SD card.

I did try using ee_defaults.txt. This is going to sound strange, but it worked exactly once. Got an IP address, was able to connect to the network. But I wasn't able to SSH to it, so I toggled SSH and rebooted. Upon rebooting, it was back to being unable to get an IP.

shantigilbert, 2021-08-18:
its in `/emuelec/configs/emuelec.conf` or `/storage/.config/emuelec/configs` you cannot access the ext4 partition on Windows, so your best bet is to SSH it to it, or use the samba shares.

Dr-KerbalMD, 2021-08-18:
Had to get a USB C to Ethernet adapter to SSH, but I was able to SSH. I wrote the password in the conf file and saved it. I escaped the $ character like the comment says I should. Still can't get an IP.

This is also interesting: after rebooting, the password is still in the config file in plain text, but the $ character is no longer escaped.

shantigilbert, 2021-08-19:
Interesting, maybe somewhere between the call the `\` is getting removed. I will need to check this more in depth, but I do not have a WiFi AP with such password.

Dr-KerbalMD, 2021-08-19:
Ok thank you! I can just use Ethernet to SSH for now. Your router might let you create a guest network with a different SSID/password, which would be a good way to try and reproduce this.

Langerz82, 2021-09-06:
If someone mods the password code it might also be useful to have the password show up as stars on the screen for privacy and a show password option, but I guess someone can follow the order if you use say a joystick and memorise.

Langerz82, 2021-09-26:
I used a wifi password with a dollar sign in the middle, obviously entering it in ES is impossible, but with a backslash and dollar sign in emuelec.conf, and restarting ES, it connected.

closed 2022-01-21 by shantigilbert