EmuELEC/EmuELEC #700: How to remove the script that creates \\ EMUELEC \ Storage \ roms \

SlowPixel, 2021-08-31:
The crux of the problem:
EmuElec 4.2 is installed as a second system on Ugoos x3.
How to remove the script that creates \\ EMUELEC \ Storage \ roms \
Why do I need this: Roms with SDCARD is not mounted until I rename roms (whatever name) and reboot the system. If you completely turn off and turn on the folder along the path is created again. (\\ EMUELEC \ Storage \ roms \)

На Русском
Суть проблемы:
EmuElec 4.2 установлена как вторая система на Ugoos x3.
Как удалить скрипт создающий \\EMUELEC\Storage\roms\ 
Для чего мне это нужно: Roms с SDCARD не монтируется пока я не переименую roms(в любое название) и не перезагружу Emuelec. Если полностью выключить и включить папка по пути создается опять.(\\EMUELEC\Storage\roms\ )

shantigilbert, 2021-08-31:
It cannot be removed

closed 2021-08-31 by shantigilbert

SlowPixel, 2021-08-31:
> It cannot be removed

Are there any solutions to this problem?

shantigilbert, 2021-08-31:
Unless translation is wrong, this is not a problem, it is working as intended, the ROMS path is a very important path in EmuELEC without it there is nothing to load!

SlowPixel, 2021-08-31:
> Unless translation is wrong, this is not a problem, it is working as intended, the ROMS path is a very important path in EmuELEC without it there is nothing to load!

And why then, after renaming and rebooting, roms start loading from SD CARD or USB? It turns out that the script does not work when Emuelec is in the internal memory of the TV BOX.

SlowPixel, 2021-08-31:
> Unless translation is wrong, this is not a problem, it is working as intended, the ROMS path is a very important path in EmuELEC without it there is nothing to load!

it works when renaming Roms for example to Rom

shantigilbert, 2021-08-31:
I do not understand what you are trying to say. Sorry. 

if you are trying to load roms from USB please read the WiKi https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-USB-media

other than that I cannot understand what the "problem" is

SlowPixel, 2021-08-31:
> I do not understand what you are trying to say. Sorry.
> 
> if you are trying to load roms from USB please read the WiKi https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-USB-media
> 
> other than that I cannot understand what the "problem" is

I recorded a video clip https://cloud.mail.ru/home/%D0%B2%D1%80%D0%B5%D0%BC%D0%B5%D0%BD%D0%BD%D0%B0%D1%8F/VID_20210831_204725.mp4

SlowPixel, 2021-08-31:
> I do not understand what you are trying to say. Sorry.
> 
> if you are trying to load roms from USB please read the WiKi https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-USB-media
> 
> other than that I cannot understand what the "problem" is

at 3.9 everything works fine
The problem started at 4.0-4.3

shantigilbert, 2021-08-31:
This is something on your end, that is not the behavior of the official image. `/storage/roms` is just a mount/symlink to either the EEROMS partition or an external rom source (like USB/SD) 

You can either reset your configuration to default by going into EmuELEC settings, Danger Zone, reset system to default to see if that helps, or do a clean install.

SlowPixel, 2021-08-31:
> This is something on your end, that is not the behavior of the official image. `/storage/roms` is just a mount/symlink to either the EEROMS partition or an external rom source (like USB/SD)
> 
> You can either reset your configuration to default by going into EmuELEC settings, Danger Zone, reset system to default to see if that helps, or do a clean official install.

the image is original, the only exception is it is sewn with the second AOSP + EmuELec system
Thanks for the answers, I will pick.