EmuELEC/EmuELEC #702: usb ports not working on UGOOS X2CUBE Box (g12a_s905x2_4g_1gbit). When i went back and boot back to android my ports are working ok

cyberazi, 2021-09-05:
**Describe the bug**
usb ports not working on UGOOS X2pro Box (g12a_s905x2_4g_1gbit). mu Elec Boot normally but there is no way to input anything. If you asking about my remote. My original IR remote died long time ago so I use a 2.4GHZ airmouse remote and keyboard

When i went back and boot back to android my usb ports are working and i can use my controler, Airmouse and my keyboard

**To Reproduce**
Steps to reproduce the behavior:
1. Just flash the firmware i tried both g12a_s905x2_4g_1gbit and g12a_s905x2_4g.dtb. will boot normally jsut cannot input anything when at the startup page

**Expected behavior**
I should be able to start conifguring my controller which work fine on my BEElink GT king unit on emuelec

**Screenshots**
none can do screen capture since no input available

**EmuELEC version:**
 - EmuELEC-Amlogic-ng.aarch64-4.1-Generic.img.gz

**Device (please complete the following information):**
 - SOC: S095x2
 - Brand: UGOOS
 - Model X2
 - Version FIRMWARE VERSION 0.4.1, HARDWARE VERSION 2

SlowPixel, 2021-09-05:
Нужно найти файл autostart.sh, который находится //emuelec/configfiles
И добавить 4 строчки
#!/bin/sh
echo 432 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio432/direction
echo 1 > /sys/class/gpio/gpio432/value
echo 432 > /sys/class/gpio/unexport

Это работает на UGOOS X3pro

shantigilbert, 2021-09-05:
You already got an answer to this: https://emuelec.discourse.group/t/ugoos-x2-cube-usb-wont-work-on-emuelec/422/2?u=emuelec

closed 2021-09-05 by shantigilbert