EmuELEC/EmuELEC #711: [EmulationStation][Wifi]Turning on and off sometimes crashes ES.

Langerz82, 2021-09-14:
**Describe the bug**
In Emulation Station turning the wifi on and off will sometimes crash ES completely. You cannot exit the settings dialog Screen or turn off the device. You have to pull out the power cable and cold boot.

**To Reproduce**
1. Go into Network settings and turn the Wifi on and off, off and on.
2. Sometimes it will completely crash ES.

**Expected behavior**
Turning wifi on and off should not crash ES.

**EmuELEC version:**
Latest, Dev

**Device (please complete the following information):**
 - SOC: S922X
 - Brand: Beelink
 - Model GT King Pro

shantigilbert, 2021-09-14:
Is this the WiFi 6 device?

Langerz82, 2021-09-15:
Yeah I believe so