EmuELEC/EmuELEC #720: Favorite not in alphabetic order in last 4.3 test

aurunmilla, 2021-09-20:
as the title, in the 4.3 test version, the games in the favorites section are not in alphabetical order, even if this has been selected.

Langerz82, 2021-09-25:
Do you in view options have sort games by "Filename, Ascending" ?
(Select) View Options: Whats the gamelist view style? Basic, Carousel or Grid? 
Also make sure: (Select) View Options >> View Customization, Show Favourites on top, choose YES.

Langerz82, 2021-09-26:
It's strange sometimes a game added to favourites gets sort corrupted and once that occurs any game you add to favourites can also sort corrupt. When disabled favourite on the games with this issue, they dont resume there normal position in the sorted list, they get added onto the end.
So far I havent been able to isolate and replicate how a game corrupts the sorting to begin with. Also restarting ES does not re-sort the gamelist alphabetically properly and the bugged sort games remain added to the end of the list regardless of the letters they start with.

Langerz82, 2021-09-26:
If you remove the game list file in /storage/roms/<emulator>/gamelist.xml. And re-add the favourites, so far it has not corrupted on me. This means you will probably have to scrape again, but it's the only solution I found so far that restores sorting order.

aurunmilla, 2021-09-26:
I think this is an issue, because if i update from 4.0 to 4.2, the sorting its seems all ok from my previous system.
So, i think is not correct to force people to re-scrap hundreds of roms as work-around.
That is an issue.

Langerz82, 2021-09-26:
I'm not sure but I think it may be to do with the name of the file, being changed to sort by sortname.
https://github.com/EmuELEC/emuelec-emulationstation/blob/67446332e837c667e0719e6f84bff0e0cdb6868c/es-app/src/FileSorts.cpp#L82

Langerz82, 2021-09-26:
I proposed a fix that provides backward compatibility if sortname cannot be compared.

aurunmilla, 2021-09-26:
Yes, also because i start to not create any gamelist.xml in any folder, so i had no one gamelist in my system.
This means that also i do not have any "SORTNAME" tab avaiable for any platform.
It shoulb be usefull a backward compatibilty fix.

Langerz82, 2021-09-27:
It's been merged on the dev branch but give it a couple days as EmuELEC needs to point to the latest EmulationStation version.

shantigilbert, 2021-09-27:
> 
> 
> It's been merged on the dev branch but give it a couple days as EmuELEC needs to point to the latest EmulationStation version.

it was reverted as there were changes to upstream, please check it

Langerz82, 2021-10-01:
You can test the changes here:
https://github.com/EmuELEC/emuelec-emulationstation/pull/35

closed 2022-03-19 by shantigilbert