EmuELEC/EmuELEC #726: Bezel project not working on new odroid super

gguy825, 2021-09-25:
Describe the bug
Installed emuelec on a new 256gb sd card and setting up bezels for consoles like PSX, gameboy, and snes/nes but using the bezel project, nothing comes up to download. The themes list populate with results but when I go to the bezel project, nothing comes up. Im connected to my wifi but still nothing. Ive tried using versions 4.1 and 4.2 but nothing comes up in the list. Even adding them manually doesn't work. Am I missing something beforehand or should I have installed something extra along with it before trying to get bezels? First handheld so still learning on how to get it up and running. 

EmuELEC version:
Latest, 4.1

Device (please complete the following information):
New odroid super using EmuELEC-OdroidGoAdvance.aarch64-4.2-odroidgo2.img.gz

shantigilbert, 2021-09-25:
The bezel project only shows depending on what ROMS you have, do you have roms?

gguy825, 2021-09-25:
> The bezel project only shows depending on what ROMS you have, do you have roms?
right now I reinstalled 4.2 and the only rooms i have on right now are Spyro year of the dragon and dead or alive

shantigilbert, 2021-09-25:
Then that is why its not showing any bezels. install ROMS then try it again.

closed 2021-09-25 by shantigilbert

gguy825, 2021-09-25:
Does it have to have a minimum number? I need to add more than 2?

shantigilbert, 2021-09-25:
No, as long as you have at least one ROM for a compatible system, not all systems support bezels.

gguy825, 2021-09-25:
Ahh ok then. Ill load more from my collection and go from there. ty

shantigilbert, 2021-09-25:
if it doesn't work feel free to re-open this issue.

gguy825, 2021-09-25:
Just to update. it is working for some systems, GBA, SNES, but not for things like sega saturn or psx. having to manually do ones for those systems. i have around 30 rooms on both saturn and psx currently. Any idea?