EmuELEC/EmuELEC #727: EMUELEC 4.2 Can't boot from MicroSD.

phantoma13, 2021-09-25:
Hi everybody
I have a x96 Max+ tv box that has s905x3 2 GB RAM Not the gigabit version
I can't boot to EMUELEC from MicroSD, I think I'm using the right DTB because the device succesfully boots if I use a USB Flash Drive instead of the MicroSD. The attached screen is the result of trying to boot from MicroSD

Could anyone help me please? I bought this device because I read that was compatible.

Thanks in advance

[image]

shantigilbert, 2021-09-25:
file:///android_asset/md/#about-cheap-devices

british-choi, 2021-09-26:
The attached screen is the result of trying to boot from MicroSD? where?

phantoma13, 2021-09-28:
> The attached screen is the result of trying to boot from MicroSD? where?

Sorry, i didn't notice i used the wrong picture.

closed 2021-09-29 by shantigilbert