EmuELEC/EmuELEC #730: 4.3 test release on s905X3 chip. reported stand-alone ScummVM with too HI graphycs pre-setting

aurunmilla, 2021-09-28:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

closed 2021-09-28 by shantigilbert

aurunmilla, 2021-09-28:
compared to version 4.2, the stand-alone version of scummvm suffers from notable slowdowns on s905X3.
Comparing it with the version I noticed that the settings on the 4.3 test, in the graphic options page, do not allow you to select any graphic filter; while on 4.2 yes.
My suspicion is that a very strong graphic filter has been put on the 4.3, and that it is not possible to change it.

shantigilbert, 2021-09-28:
There is no filter, this has been discussed before, emulators go for accuracy, not performance, Scummvm is the same. Other than reverting the version, there is no magical fix that I know of, unless you ask them to consider slower SOCS.

aurunmilla, 2021-09-28:
I would not have opened a discussion on this question if I had not seen stand-alone scummVM running very well on the 4.2 on the s905X3.

shantigilbert, 2021-09-28:
Sure... but are you missing my point?

aurunmilla, 2021-09-28:
I raise my hands, I just gave feedback