EmuELEC/EmuELEC #742: Khadas problems

jv84, 2021-10-18:
I can't boot EmuElec from SD card on Khadas VIM3L.
I have copied and renamed the DTB file. CoreELEC image works fine.

Khadas have released EmuElec image by themselves but it's two years old.
I tried to update from the system but the root partition is too small.

jv84, 2021-10-30:
I don't understand what template should I follow. It seems that you're not very helpful with this issue so I'll just close it.

closed 2021-10-30 by jv84

shantigilbert, 2021-10-30:
You don't understand because you blindly open the issue, its literally the first screen you get right after you click on "New Issue"  you even had to delete all the info that appears when opening the issue, again it seems without even reading it. 

But you are also right, I won't be helpful with this issue as I do not have a VIM3L and I have no idea how to fix this "issue".