EmuELEC/EmuELEC #757: Could you teach me how to set up the joystick of Game cube on Beelink gt king pro?

title666, 2021-10-30:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

title666, 2021-10-30:
Could you teach me how to set up the joystick of Game cube on Beelink gt king pro?

AlienatedDiego, 2021-10-30:
https://emuelec.discourse.group/t/set-up-joystick-gamecube/628

closed 2021-10-30 by shantigilbert