EmuELEC/EmuELEC #759: Overclock on Odroid Go Advance/Super does not work

francistheodorecatte, 2021-11-08:
**Describe the bug**
Overclocking to 1.4 or 1.5GHz via the DANGER ZONE menu in ES, or via setting the scaling_governor, scaling_max_freq, and scaling_current_freq manually over SSH, does not work.

**To Reproduce**
Toggle OVERCLOCK inside the Emuelec DANGER ZONE menu from OFF, to 1.4GHZ, or 1.5GHZ.

or, via SSH, run the following commands:
echo userspace > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
echo 1416000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed
echo 1416000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq

**Expected behavior**
/sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq should be set to 1416000, or 1512000, but is not; it stays at 1296000.

**EmuELEC version:**
 - Version 4.3

**Device (please complete the following information):**
 - SOC: RK3326
 - Brand: Hardkernel
 - Model: Odroid-Go Super

**Additional context**
/sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies only shows the below frequencies, regardless of the governor:
408000 600000 1008000 1200000 1296000

shantigilbert, 2021-11-09:
I think (can't remember right now) OC was removed because it caused so many issues, but the menu option is still there, I will check it when I have time and if it was in fact removed I will also remove the menu option.

shantigilbert, 2022-03-09:
This was removed as it only caused issues.

closed 2022-03-09 by shantigilbert