EmuELEC/EmuELEC #760: Chiquita-OGA theme no installable.

SuperBadger77, 2021-11-08:
**Describe the bug**
The Chiquita-OGA theme is not installable. I get an error every time I try to install it on an Odroid Go Super.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Updates and Downloads in Emulation Station on an OGS.
2. Click on Themes
3. Scroll down to Chiquita-OGA
4. Try and install - will fail.

The github page listed in themes is https://github.com/Squinz/Chiquita-OGA/ which gives a 404.

**Expected behavior**
Theme should install.

**Screenshots**
NA

**EmuELEC version:**
4.3

**Device (please complete the following information):**
Odroid Go Super
**Additional context**

shantigilbert, 2021-11-08:
> which gives a 404.

Nothing I can do about that, guess the author deleted it.

closed 2021-11-08 by shantigilbert

SuperBadger77, 2021-11-09:
@shantigilbert if it's been removed by the user, shouldn't it be removed from the themes list as an option to download?

shantigilbert, 2021-11-09:
@SuperBadger77 read my comment again.

SuperBadger77, 2021-11-09:
> @SuperBadger77 read my comment again.

Apologies - I misread . Just tested it and it's gone. Thank you!