EmuELEC/EmuELEC #761: HELP ME TO CHANGE LOGO EMUELEC

linhduonghg, 2021-11-09:
ANYBODY HERE CAN HELP ME HOW TO CHANGE LOGO EMUELEC WHEN STARUP (NOT SPLASH VIDEO), I WANT TO CHANGE LOGO EMUELEC TO SOMETHING ELSE THAT MADE BY MYSELFT, I FOUND EVERYWHERE BUT NOTHING. TKS SO MUCH. HOPE THIS ISSUE IS NOT DELETED.

AlienatedDiego, 2021-11-09:
It will surely be deleted and you already know it, it is not an EmuELEC issue.
afaik the startup logo cannot be changed unless you compile your own image.
 if there is an easier way to do it it would be a good thing if the dev share it in my opinion,
all those modified images, .. at least they would not discredit the name of emuelec

linhduonghg, 2021-11-09:
> It will surely be deleted and you already know it, it is not an EmuELEC issue. afaik the startup logo cannot be changed unless you compile your own image. if there is an easier way to do it it would [...]

tks for your reply

closed 2021-11-09 by shantigilbert