EmuELEC/EmuELEC #763: AI game translate service_mode configuration has bug in emuelec 4.3.

soominpapa, 2021-11-10:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
There is a bug in the AI ​​game translation function in Emuelec 4.3.

first bug,
In Emu-Elec's AI game translation function menu, the Target Language setting value is set as the source value in Retro Arch. Emu-Elec's settings are set incorrectly.

second bug,
The translation output of the AI ​​game translation function is fixed in voice mode.
Even if I modify it to image mode in retroarch's retroarch.cfg file, the setting value is changed to voice mode again.
My guess is that there should be an option in the AI ​​game mode settings menu for Emu-Elec to make a choice for the translation output.
Since there is no option for it in Emu-Elec, it seems to be fixed only in voice mode even if I modify it in Retroarch.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version 4.3

**Device (please complete the following information):**
 - SOC: 
 - Brand : odroid
 - Model : odroid N2
 - Version

**Additional context**
Add any other context about the problem here.

shantigilbert, 2021-11-10:
Can you describe step by step how to reproduce, (including what system and game you used) AFAIK translation AI was not even working at all.

soominpapa, 2021-11-11:
> Can you describe step by step how to reproduce, (including what system and game you used) AFAIK translation AI was not even working at all.

I'm not used to posting on github.
So, I posted pictures and explanations about the AI ​​game translation bug on my blog.
If you look at that post, you can reproduce the bug.
Thank you.

https://blog.daum.net/beckhom/5960546

shantigilbert, 2021-11-11:
I thought you meant you got it working on Retroarch the way you want it, but after reading your blog post, I can see that is not the case, according to your blog

>Bugs in the AI ​​game translation feature are an old problem. 

So it has never worked for you in text mode? This is what I mean on it not working at all, I have no idea how to fix it I tried, but it seems it is out of my league.

closed 2021-11-16 by shantigilbert