EmuELEC/EmuELEC #765: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.

pjessop, 2021-11-12:
Hi, I've re-installed Ubuntu on Ubuntu 20.04.3 LTS and have been struggling with building EmuELEC.

Initially, I had trouble downloading the repo and was getting a `RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.` from just git clone. I tried increasing postBuffer..etc but no luck. I therefore ran the git clone command with  `git clone https://github.com/shantigilbert/EmuELEC.git --depth 1` to get the base level and then downloaded the zip of EmuELEC from EmuELEC on the github website. It's not ideal because I can't checkout Amster this way, but I can still 'make image'.

Alas, the same problem just resurfaces again at package 244. I didn't have this problem when running Ubuntu 21.10 (but re-installed to this version as was having issues with other projects that I am working on).

Would be grateful if anyone can help. I am running Ubuntu 20.04 with Ubuntu Desktop installed. Many Thanks

Error section is..

```
INSTALL      SDL2_mixer (target)
[243/488] [DONE] install SDL2_mixer
GET      Crystal (git)
    DELETE      (/home/paul/EmuELEC/sources/Crystal/Crystal-*)
    GIT CLONE      Crystal
Cloning into '/home/paul/EmuELEC/sources/Crystal/Crystal-e7d7439b376a0e9e52a01b94d83a7479083b2fbe'...
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
FAILURE: scripts/install Crystal has failed!
[244/488] [FAIL] install Crystal
```

pjessop, 2021-11-12:
..update

I was able to get around above error by copying the '/sources/Crystal' folder from a previously successful build on Ubuntu 21.10 & this took me to the retroarch build which failed with the same error, but the same tactic worked on copying the folder from a previous successful build.

I've now flown through to 321/488, where there's a new error that I can't get past. Any help would be much appreciated:

```
    TOOLCHAIN      cmake (auto-detect)
Executing (host): cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/etc/cmake-x86_64-linux-gnu.conf -DCMAKE_INSTALL_PREFIX=/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain -DSDL12TESTS=off /home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/sdl12-compat-c5bd71de527b981ec736dbd94e973cc11de0cbc0
-- Could NOT find SDL2 (missing: SDL2_LIBRARY SDL2_INCLUDE_DIR) 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/sdl12-compat-c5bd71de527b981ec736dbd94e973cc11de0cbc0/SDL2_INCLUDE_DIR
   used as include directory in directory /home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/sdl12-compat-c5bd71de527b981ec736dbd94e973cc11de0cbc0

-- Configuring incomplete, errors occurred!
See also "/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/sdl12-compat-c5bd71de527b981ec736dbd94e973cc11de0cbc0/.x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
See also "/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/sdl12-compat-c5bd71de527b981ec736dbd94e973cc11de0cbc0/.x86_64-linux-gnu/CMakeFiles/CMakeError.log".
FAILURE: scripts/build sdl12-compat:host has failed!
[321/488] [FAIL] build   sdl12-compat:host
```

pjessop, 2021-11-15:
@shantigilbert @miwasp pls do you have any suggestions on this? I’m so keen to get this over the line so that I can perhaps help support going forward 🙏

miwasp, 2021-11-15:
I‘m also on Ubuntu 20.04.3 LTS and never had any of those errors. I‘m pretty sure there is nothing that can be done in the EmuELEC repo to fix those errors. I would try to first solve the initial error while downloading the repo. A quick search shows a wide range of root causes and solutions, from simply rebooting, missing packages like openssl to parental controls on the router and so on.
I will paste lists of installed packages related to tls and openssl on my system. Maybe this helps you identifying whats wrong.

```
# apt search tls | grep installed

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

cadaver/focal,now 0.23.3-2.1build1 amd64 [installed]
libcurl3-gnutls/focal-updates,focal-security,now 7.68.0-1ubuntu2.7 amd64 [installed]
libdevmapper1.02.1/focal,now 2:1.02.167-1ubuntu1 amd64 [installed]
libevdev2/focal-updates,now 1.9.0+dfsg-1ubuntu0.1 amd64 [installed,automatic]
libfstrm0/focal,now 0.6.0-1 amd64 [installed,automatic]
libgnutls26/now 2.12.23-12ubuntu2.5 amd64 [installed,local]
libgnutls30/focal-updates,focal-security,now 3.6.13-2ubuntu1.6 amd64 [installed,automatic]
libksba8/focal,now 1.3.5-2 amd64 [installed,automatic]
liblwp-protocol-https-perl/focal,focal,now 6.07-2ubuntu2 all [installed,automatic]
libneon27-gnutls/focal,now 0.30.2-4 amd64 [installed,automatic]
libnet-smtp-ssl-perl/focal,focal,now 1.04-1 all [installed,automatic]
libnss3/focal-updates,focal-security,now 2:3.49.1-1ubuntu1.5 amd64 [installed,automatic]
libsoup-gnome2.4-1/focal,now 2.70.0-1 amd64 [installed,automatic]
libsoup2.4-1/focal,now 2.70.0-1 amd64 [installed,automatic]
libssl-dev/focal-updates,now 1.1.1f-1ubuntu2.9 amd64 [installed]
libssl1.1/focal-updates,now 1.1.1f-1ubuntu2.9 amd64 [installed,automatic]
openssl/focal-updates,now 1.1.1f-1ubuntu2.9 amd64 [installed]
python3-certifi/focal,focal,now 2019.11.28-1 all [installed,automatic]
rsyslog/focal-updates,now 8.2001.0-1ubuntu1.1 amd64 [installed]
```

```
# apt search openssl | grep installed

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libapache2-mod-php7.4/focal-updates,focal-security,now 7.4.3-4ubuntu2.7 amd64 [installed,automatic]
libcurl4/focal-updates,focal-security,now 7.68.0-1ubuntu2.7 amd64 [installed,automatic]
libssh-4/focal-updates,focal-security,now 0.9.3-2ubuntu2.2 amd64 [installed,automatic]
libssl-dev/focal-updates,now 1.1.1f-1ubuntu2.9 amd64 [installed]
libssl1.1/focal-updates,now 1.1.1f-1ubuntu2.9 amd64 [installed,automatic]
openssl/focal-updates,now 1.1.1f-1ubuntu2.9 amd64 [installed]
perl-openssl-defaults/focal,now 4 amd64 [installed,automatic]
php7.4-cgi/focal-updates,focal-security,now 7.4.3-4ubuntu2.7 amd64 [installed,automatic]
php7.4-cli/focal-updates,focal-security,now 7.4.3-4ubuntu2.7 amd64 [installed,automatic]
python3-openssl/focal,focal,now 19.0.0-1build1 all [installed,automatic]
python3-service-identity/focal,focal,now 18.1.0-5build1 all [installed,automatic]
ssl-cert/focal,focal,now 1.0.39 all [installed]
```

shantigilbert, 2021-11-16:
@pjessop 

> 
> 
> @shantigilbert @miwasp pls do you have any suggestions on this? I’m so keen to get this over the line so that I can perhaps help support going forward 🙏

I think @miwasp is right that this has nothing to do with the repo itself. 

> I‘m also on Ubuntu 20.04.3 LTS and never had any of those errors. I‘m pretty sure there is nothing that can be done in the EmuELEC repo to fix those errors. I would try to first solve the initial [...]

I did a quick google search (which I am sure you did as well, but just in case) and found this <https://stackoverflow.com/questions/38378914/how-to-fix-git-error-rpc-failed-curl-56-gnutls> maybe try one of the solutions there?

ws02589111, 2021-11-16:
@shantigilbert 
Hi，I found the `pcre` and `pcre2` pkg repo url now is unavailable when I build the dev. 
according to the pcre offical site, we may change it from
`https://ftp.pcre.org/pub/pcre/`
to
`https://sourceforge.net/projects/pcre/files/`

This is my changes
pcre: `PKG_URL="https://jaist.dl.sourceforge.net/project/pcre/pcre/$PKG_VERSION/$PKG_NAME-$PKG_VERSION.tar.bz2"`
pcre2: `PKG_URL="https://jaist.dl.sourceforge.net/project/pcre/pcre2/$PKG_VERSION/$PKG_NAME-$PKG_VERSION.tar.bz2"`

pjessop, 2021-11-16:
Many thanks @miwasp @shantigilbert for coming back with some suggestions.

I have now worked out that the issue with `RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.` is actually my broadband internet provider (Virgin Media in this case). If I use my internet provider via ethernet or wifi then I get this error. 

However, if I disconnect from my internet provider and tether to my phone's data, then I can download the repo (which is my only other means of pulling data). 

It means I'm stuffed for phone data for the rest of the month, but at least i got EmuELEC downloaded :)

I've just started the build, but of course have had to switch back to the problematic internet provider. I'll no doubt get stuck on crystal, retroarch, mrboom and ppsspp packages (as those failed on the same error where the fudge mentioned in original post got me as far as ppsspp). However, I guess when I error out on these, I can switch back to my phone to pull required content of these packages, then cut the connection after they successfully build and force an error out on the next package (as I can't build everything from my phone data). 

Of course, the problem will be if crystal, retroarch, mrboom, ppsspp and any remaining packages (after 372) that cause this error, are pulling large levels of data remotely in which case my phone data won't handle it. With a bit of luck (and I'm due some luck on trying to compile this image :)), I can limp over the line and get onto the exciting stuff!!

shantigilbert, 2021-11-17:
> @shantigilbert Hi，I found the `pcre` and `pcre2` pkg repo url now is unavailable when I build the dev. according to the pcre offical site, we may change it from `https://ftp.pcre.org/pub/pcre/` to [...]

https://github.com/EmuELEC/EmuELEC/commit/6c1fd4165c6152d5abcb81b000932f6c3cb2cc1c

shantigilbert, 2021-11-17:
> Many thanks @miwasp @shantigilbert for coming back with some suggestions.
> 
> I have now worked out that the issue with `RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.` is actually my broadband internet provider (Virgin Media in this case). If I use my internet provider via ethernet or wifi then I get this error.
> 
> However, if I disconnect from my internet provider and tether to my phone's data, then I can download the repo (which is my only other means of pulling data).
> 
> It means I'm stuffed for phone data for the rest of the month, but at least i got EmuELEC downloaded :)
> 
> I've just started the build, but of course have had to switch back to the problematic internet provider. I'll no doubt get stuck on crystal, retroarch, mrboom and ppsspp packages (as those failed on the same error where the fudge mentioned in original post got me as far as ppsspp). However, I guess when I error out on these, I can switch back to my phone to pull the remaining required content of these packages, then cut the connection after they successfully build and force an error out on the next package (as I can't build everything from my phone data), reconnect my virgin media internet and sprint back to where I was from `make image` again. That's the plan!! :)
> 
> Of course, the problem will be if crystal, retroarch, mrboom, ppsspp and any remaining packages (after 372) that cause this error, are pulling large levels of data remotely in which case my phone data won't handle it. With a bit of luck (and I'm due some luck on trying to compile this image :)), I can limp over the line and get onto the exciting stuff!!

Glad you found the issue, I am surprised by the reason tho, did nothing from the link I posted worked for you? seems very weird that your ISP would be blocking this somehow.

pjessop, 2021-11-17:
@shantigilbert well, it turned out to be a coincidence and you are right that the ISP perhaps isn't the culprit.

I used my strategy above of using phone data when I got to the problem packages, but they just gave the same error message again. I therefore, as I did before, tried copying over the Ubuntu 21.10 folders from /sources on the packages where this error killed the build, which worked for Crystal, Retroarch, mrboom, but the fudge won't get me past ppsspp because that package is updated on Ubuntu 21.10 and needs libzip and libpng 1.7 and the highest I can upgrade those packages on in Ubuntu 20.04 is 1.6. It errors out on that reason at 372/488.

I've tried everything on the various articles for fixing the RPC failed error but nothing seems to work. So frustrating :(

I'm not sure whether to go back to Ubuntu 21.10 and install on a bigger drive as I did make it to the end of that build (but ran out of memory). I am determined to get this working and support if I can :)

pjessop, 2021-11-17:
@shantigilbert @miwasp the issue has been resolved.

It turned out the Child Lock on the ISP was causing this issue. I'm not sure how & why, but it now works.

It means my kids can download porn and I can compile EmuElec. Everyone is happy!! :-)  lol lol

pjessop, 2021-11-17:
@shantigilbert @miwasp sorry, I spoke too soon!! Not quite at the end of the build yet. A new problem this time it seems and looks like some issue with SDL. Any clues in the log..
```

INSTALL      np2kai (target)
[430/486] [DONE] install np2kai
BUILD      jzintv (target)
    TOOLCHAIN      make
Executing (target): make -C src/ -f Makefile GNU_READLINE=0 
make[1]: Entering directory '/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/jzintv-20200712/src'
make[1]: /home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin/sdl2-config: Command not found
make[1]: /home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin//home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/bin/sdl2-config: Command not found
/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/bin/aarch64-libreelec-linux-gnueabi-gcc -c -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a73.cortex-a53 -mcpu=cortex-a73.cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -O3  -flto=auto -ffat-lto-objects    -I. -I..    -DUSE_SDL2   -DJZINTV_VERSION_MAJOR=0x01 -DJZINTV_VERSION_MINOR=0x00 -DJZINTV_SVN_REV= -DJZINTV_SVN_DTY= -o svn_revision.o svn_revision.c
/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++ -o ../bin/jzintv jzintv.o pads/pads_intv2pc.o cfg/cfg.o cfg/usage.o cfg/mapping.o misc/crc32.o misc/crc16.o misc/avl.o misc/ll.o misc/file_crc32.o misc/types.o misc/jzprint.o misc/printer.o plat/plat_lib.o plat/gnu_getopt.o plat/plat_gen.o event/event.o event/event_tbl.o gfx/palette.o gfx/gfx.o mvi/mvi.o debug/debug.o debug/debug_dasm1600.o util/symtab.o debug/source.o periph/periph.o cp1600/cp1600.o cp1600/op_decode.o cp1600/op_exec.o cp1600/emu_link.o cp1600/op_exec_ext.o cp1600/req_q.o cp1600/tbl/fn_cond_br.o cp1600/tbl/fn_dir_2op.o cp1600/tbl/fn_imm_2op.o cp1600/tbl/fn_impl_1op_a.o cp1600/tbl/fn_impl_1op_b.o cp1600/tbl/fn_ind_2op.o cp1600/tbl/fn_reg_1op.o cp1600/tbl/fn_reg_2op.o cp1600/tbl/fn_rot_1op.o cp1600/tbl/formats.o mem/mem.o icart/icart.o icart/icartrom.o icart/icartbin.o icart/icartfile.o icart/icarttag.o icart/icarttag_authors.o icart/icarttag_printer.o stic/stic.o  pads/pads.o pads/pads_cgc.o ay8910/ay8910.o ecs/ecs.o ivoice/ivoice.o speed/speed.o minilzo/minilzo.o file/file.o file/elfi.o lzoe/lzoe.o metadata/metadata.o metadata/cfgvar_metadata.o metadata/icarttag_metadata.o metadata/print_metadata.o bincfg/bincfg.o bincfg/bincfg_grmr.tab.o bincfg/bincfg_lex.o  bincfg/legacy.o gif/gif_enc.o gif/lzw_enc.o demo/demo.o  name/name.o name/name_list.o serializer/serializer.o jlp/jlp.o scale/scale2x.o scale/scale3x.o locutus/locutus.o locutus/luigi.o locutus/locutus_adapt.o locutus/locutus_types.o locutus/loc_to_bin.o locutus/bin_to_loc.o locutus/rom_to_loc.o zlib/deflate.o zlib/adler32.o zlib/trees.o zlib/zutil.o avi/avi.o cheat/cheat.o svn_revision.o plat/plat_sdl.o plat/main_sdl.o event/event_sdl.o event/event_sdl2.o gfx/gfx_sdl2.o gfx/gfx_scale.o gfx/gfx_prescale.o snd/snd_sdl.o joy/joy_sdl.o mouse/mouse_sdl.o -march=armv8-a+crc+crypto+fp+simd -mabi=lp64 -Wno-psabi -mtune=cortex-a73.cortex-a53 -mcpu=cortex-a73.cortex-a53+crc+crypto+fp+simd -fomit-frame-pointer -Wall -pipe -O3  -O3  -flto=auto -ffat-lto-objects    -I. -I..    -DUSE_SDL2   -DJZINTV_VERSION_MAJOR=0x01 -DJZINTV_VERSION_MINOR=0x00 -DJZINTV_SVN_REV= -DJZINTV_SVN_DTY=  -lrt 
/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/lib/gcc/aarch64-libreelec-linux-gnueabi/8.3.0/../../../../aarch64-libreelec-linux-gnueabi/bin/ld: /tmp/ccIy96vG.ltrans4.ltrans.o: in function `plat_quit':
<artificial>:(.text+0x20e0): undefined reference to `SDL_Quit'
/home/paul/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4.3/toolchain/lib/gcc/aarch64-libreelec-linux-gnueabi/8.3.0/../../../../aarch64-libreelec-linux-gnueabi/bin/ld: /tmp/ccIy96vG.ltrans4.ltrans.o: in function `gfx_toggle_windowed':
<artificial>:(.text+0x5750): undefined reference to `SDL_SetWindowFullscreen'
[... cut ...]
parallel: This job failed:
package_worker 6 432 486 'install jzintv'
Parallel build failure - see log for details. Time of failure: Wed Nov 17 15:43:53 GMT 2021
make: *** [Makefile:12: image] Error 1

```
Here's what I have installed for libsdl2

```
paul@paul-desktop:~/EmuELEC$ apt list libsdl2-dev
Listing... Done
libsdl2-dev/focal,now 2.0.10+dfsg1-3 amd64 [installed]
libsdl2-dev/focal 2.0.10+dfsg1-3 i386
```
Would really appreciate your help to try to get over this final(ish) hurdle!!

Many Thanks

ws02589111, 2021-11-18:
> I‘m also on Ubuntu 20.04.3 LTS and never had any of those errors. I‘m pretty sure there is nothing that can be done in the EmuELEC repo to fix those errors. I would try to first solve the initial [...]

what version is your gcc and cmake？

miwasp, 2021-11-18:
> @miwasp what version is your gcc and cmake?

Standard packages of Ubuntu 20.04.
```
$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cmake --version
cmake version 3.16.3
```

ws02589111, 2021-11-18:
> Standard packages of Ubuntu 20.04.
>
> ```
> $ gcc --version
> gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
> Copyright (C) 2019 Free Software Foundation, Inc.
> This is free software; see the source for copying [...]

OK, thx. yours gcc and cmake version is lower than me.
I updated my ubuntu from 18.04 Desktop to 20.04.3 Server, but get many complie error in some packages(with lower version gcc and cmake, the errors solved but new error get).
I will retry complie with yours tools version

pjessop, 2021-11-18:
@shantigilbert many thanks the clean and build worked. It errored out near the end on pcre2 as it couldn't pull from the url, but I noticed you referenced a new url for that, so all good.

A huge thanks to you and @miwasp for getting me through the build.

I'm through the build and successfully flashed to an SD and now working on an Amlogic box.

Do you have any developer docs or discord where developers can learn more? I'd be keen to contribute, but this repo is huge and I'd need to get a grasp of it and understand more first.

thanks again.

shantigilbert, 2021-11-18:
> Do you have any developer docs or discord where developers can learn more? I'd be keen to contribute, but this repo is huge and I'd need to get a grasp of it and understand more first.

There is a dev channel on the EmuELEC discord (https://discord.gg/cbgtJTu) there is also a forum at https://emuelec.org but not much dev discussion there, and there is some minor info on the wiki.

I am closing this issue now as it has been resolved.

closed 2021-11-18 by shantigilbert