EmuELEC/EmuELEC #767: some systems do not respect global resolution

AlienatedDiego, 2021-11-14:
I have changed the global resolution to 720p, after the system restarts I observe in game settings / per system advanced conf that some systems have effectively changed to 720p, but others keep the 1080p resolution option checked. I mean by this: it seems not all systems are maintaining the resolution requested in the global options (emuelec settings)

if the resolution per system is set to “auto” it DOES respect the global resolution. but “auto” is not the default option, it must be set manually for each system.

then I would like to request a possible check / change to establish the “auto” option by default in the resolution values ​​of the systems in ES so that in this way the global resolution is the starting point and each one can establish the independent resolution that want per system.

Thank you for your attention in this matter.

shantigilbert, 2021-11-14:
When you say some games, what games do you refer to? What systems, etc you missed this and it's very important.

shantigilbert, 2021-11-14:
@Langerz82 can you check this one?

AlienatedDiego, 2021-11-14:
device: Tanix tx3 (s905x3) with external hdd, system on sd  
EmuELEC version: latest compilable test version.

how to reproduce:  
.We set EmuELEC by default and check game settings / advanced options by emulator / systems  
. We check that in all systems all the options are set to "auto", EXCEPT resolution: set to "1080p60hz"  
. Let's go to the main menu to emuelec settings and change the global resolution of the system from 1080p60hz to 720p60hz  
Let's go again to game settings / advanced options by emulator and check all systems / resolution.  
. we note that some have set to "720p60hz" and others to "1080p60hz" (THIS IS RANDOM, not game / system specific)

how it should work:  
system resolutions should be set to "auto" to correctly inherit the global resolution set, or failing that, the same set as global in emuelec settings

note: resolutions set by system / game (whatever they are) are displayed correctly

Langerz82, 2021-11-14:
Simple fix for it:
https://github.com/EmuELEC/emuelec-emulationstation/pull/36

I have not tested. @AlienatedDiego can you test for me? Thanks.

AlienatedDiego, 2021-11-14:
I'll check it asaip. of course only you and Shantigilbert decide to keep the change or not.Thank you very much

Langerz82, 2021-11-14:
It's cool I think this change is for the better, "auto" option was a change added later and it should be implemented in properly which this should fix.

closed 2022-03-19 by shantigilbert