EmuELEC/EmuELEC #770: Change resolution black screen

keeno1982, 2021-11-25:
**Describe the bug**
Change resolution black screen

**To Reproduce**
Steps to reproduce the behavior:
Change emuelec resolution to 1024x768,1280x1024,1600x1200 black screen

**Expected behavior**
Display normal

**Screenshots**

**EmuELEC version:**
V4.3

**Device (please complete the following information):**
 - SOC: S922x
 - Brand: Beelink
 - Model: GT-King pro
 - Version

**Additional context**
Monitor: Dell 2007fpb, resolution 1600x1200
Connect: HDMI-DVI

I love emuelec and retro game, thanks for you. I hope you can help me to solve this problem, I had try many ways and search on google many days and the issue still here. Please don't close issue quickly, I hope it can fix if it's a bug or you help me find a way to  
deal it, thanks.

shantigilbert, 2021-11-26:
https://github.com/EmuELEC/EmuELEC/wiki/Common-Questions---FAQ#how-to-change-resolution-i-get-black-screen-or-my-tv-is-not-working-with-your-image

keeno1982, 2021-11-26:
@shantigilbert Very sorry for this and thanks for you reply, I will try. :)