EmuELEC/EmuELEC #771: Ubuntu 20.04 Could NOT find X11 (missing: X11_X11_INCLUDE_PATH X11_X11_LIB)

peterwillcn, 2021-11-26:
ubuntu 20.04

```
-- Found Threads: TRUE
CMAKE_SYSTEM_PROCESSOR: aarch64
Generating for ARMv8, MinSizeRel
CMake Error at ext/CMakeLists.txt:3 (add_subdirectory):
  The source directory

    /home/emuelec/EmuELEC/build.EmuELEC-OdroidGoAdvance.aarch64-4.3/ppsspp-9de9420878df6805a1db40ede7bd264499cb1425/ext/armips

  does not contain a CMakeLists.txt file.

CMake Error at ext/CMakeLists.txt:18 (add_subdirectory):
  The source directory

    /home/emuelec/EmuELEC/build.EmuELEC-OdroidGoAdvance.aarch64-4.3/ppsspp-9de9420878df6805a1db40ede7bd264499cb1425/ext/glslang

  does not contain a CMakeLists.txt file.

-- Could NOT find Snappy (missing: SNAPPY_LIBRARIES SNAPPY_INCLUDE_DIR)
CMake Error at ext/CMakeLists.txt:25 (add_subdirectory):
  add_subdirectory given source "zstd/build/cmake" which is not an existing
  directory.

-- Could NOT find X11 (missing: X11_X11_INCLUDE_PATH X11_X11_LIB)
```

peterwillcn, 2021-11-26:
https://github.com/hrydgard/ppsspp/issues/9032
https://github.com/hrydgard/ppsspp/pull/10413

shantigilbert, 2021-11-26:
X11 is not used in EmuELEC at all, so this is something on your end, but sadly I do not know what, nor what you could try.

closed 2021-12-10 by shantigilbert