EmuELEC/EmuELEC #776: [BUG] IP address in footer does not update until you close and reopen the menu

gingerbeardman, 2021-12-18:
### Describe the bug
IP address in footer does not update until you close and reopen the menu

### To Reproduce
Steps to reproduce the behavior:
1. Press start to open menu
2. Choose Network settings
3. Switch on wifi
4. look at IP address in footer, it doesn't change

### Workaround
1. press start to close menu
2. press start to open menu

### Informations
 - LE Version: 4.3
 - Hardware Platform: Gameforce Chi

### Log file

### Additional context

shantigilbert, 2021-12-18:
Not a bug, this is expected.

closed 2021-12-18 by shantigilbert

gingerbeardman, 2021-12-19:
Then I would say this is misleading as it makes it seem like wifi toggle/connect isn't working.

shantigilbert, 2021-12-19:
You can say whatever you like, nowhere is this promised as a feature, so nobody is "misleading" anyone. 

The fact is that not many people are connecting and disconnecting their network just to see if the IP changes, it takes 2 button presses, or less than 1 second to refresh the screen, if you think this is wrong or you don't like it then I welcome PRs with your code to refresh screen every time something in the background happens.

gingerbeardman, 2021-12-19:
It's not a big deal. Take it easy

gingerbeardman, 2021-12-19:
BTW I meant the out-of-date information in the display was misleading. I wasn't referring to any person, so please don't take it that way.

shantigilbert, 2021-12-19:
No problem, all good 👍