EmuELEC/EmuELEC #779: [BUG]

gingerbeardman, 2021-12-19:
### Describe the bug

Notification of update even though I'm on 4.3 stable. 

### To Reproduce
Steps to reproduce the behavior:
1. Update to 4.3 stable
2. Power cycle device
3. Update available notification is displayed 

### Informations
 - LE Version: 4.3 stable
 - Hardware Platform: CHI

### Additional context

I did a second update just in case something has changed, but the banner is still being displayed afterwards.

shantigilbert, 2021-12-19:
I will take a look at this, this was supposed to be fixed and nobody else has reported it, so not sure what is going on.

gingerbeardman, 2021-12-20:
Video: https://imgur.com/MZcr6UR

countbuggula, 2021-12-20:
I'm seeing this happen too, on my N2+

linkous8-2, 2021-12-21:
Ditto for OGA-BE

jpsutton, 2022-01-19:
Also observed on Odroid C4.

snakect77, 2022-01-31:
Idem amlogic s903

shantigilbert, 2022-02-16:
This bug has been confirmed and its fixed on dev.

closed 2022-03-19 by shantigilbert