EmuELEC/EmuELEC #780: [BUG] WiFi toggle incorrect after restart

gingerbeardman, 2021-12-21:
### Describe the bug

WiFi toggle incorrect after restart

### To Reproduce
Steps to reproduce the behavior:
1. Toggle wifi off
2. Use system until it powers off
3. Charge and Power back on
4. Check network settings

### Expected
- Toggle should be off
- WiFi should be off

### Actual
- Toggle is off
- WiFi is on

### Informations
 - LE Version: 4.3 stable
 - Hardware Platform: CHI

gingerbeardman, 2021-12-25:
WiFi also switches itself back on in other circumstances.

Langerz82, 2022-01-23:
Might need the following line:
```
SystemConf::getInstance()->saveSystemConf();
```
after the lines that contain:
```
SystemConf::getInstance()->setBool("wifi.enabled", wifienabled);
```
So it saves immediately to the file.
I might create a mini patch after some other changes have been pushed.

gingerbeardman, 2022-01-23:
I found that WiFi was on inside RetroArch by default, I switched it off there and all seems well.

It would be ideal if the RetroArch setting synced with the ES system setting.

shantigilbert, 2022-02-16:
> It would be ideal if the RetroArch setting synced with the ES system setting.

This seems like something that should be doable. I'll see what I can do.

closed 2024-09-20 by Langerz82