EmuELEC/EmuELEC #782: [BUG]

groovey2k, 2021-12-30:
When using Odroid Go Advance with Retrorun/Flycast there is no possibility to enter a configuration menu for setting key mapping or transparent sorting (Per Triangle/Per Strip).

### Steps to reproduce

- using Emulation Station choose Rom and hold A button
- configure to use RETRORUN/FLYCAST as emulator in Advanced settings
- no configuration options other than "RGA Scaling" and "Enable Max Performace" appear directly in Emulation Station
- after starting Rom no hotkeys are available to configure emulator

### Expected behaviour

- either in Emulation Station or via Hotkey user is able to configure key mapping and and transparent sorting as well as further options of emulator

### Informations
 - EMUELEC Version 4.3
 - Hardware Platform: Odroid Go Advance Rev. 1.1

SamuEDL, 2021-12-30:
Yesterday they answered you here: https://github.com/EmuELEC/EmuELEC/issues/781

groovey2k, 2021-12-30:
Sorry, I thought I did a mistake!
Thanks for the information.

closed 2021-12-30 by groovey2k