EmuELEC/EmuELEC #783: ipega pg 9099 not pairing properly

dswann2, 2021-12-31:
will not pair with the emulation station bluetooth script it detects the pad and then say pairing failed so i ssh in to emuelec with putty to manual pair here is the output of the commands 

 [bluetooth]# default-agent
Default agent request successful
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# discoverable on
Changing discoverable on succeeded
[CHG] Controller 43:39:00:00:1F:AC Discoverable: yes
[bluetooth]# pairable on
Changing pairable on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 43:39:00:00:1F:AC Discovering: yes
[NEW] Device 00:1A:7D:DA:71:13 GAMING
[CHG] Device 20:20:04:17:2A:C1 Connected: yes
[NEW] Primary Service
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0008
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[NEW] Primary Service
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0009
        0000180a-0000-1000-8000-00805f9b34fb
        Device Information
[4:15 AM] Quantumslip: NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0009/char000a
        00002a29-0000-1000-8000-00805f9b34fb
        Manufacturer Name String
[NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0009/char000c
        00002a26-0000-1000-8000-00805f9b34fb
        Firmware Revision String
[NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0009/char000e
        00002a50-0000-1000-8000-00805f9b34fb
        PnP ID
[NEW] Primary Service
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0010
        91680001-1111-6666-8888-0123456789ab
        Vendor specific
[NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0010/char0011
        91680002-1111-6666-8888-0123456789ab
        Vendor specific
[NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0010/char0013
        91680003-1111-6666-8888-0123456789ab
        Vendor specific
[NEW] Descriptor
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0010/char0013/desc0015
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Primary Service
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029
        f000ffc0-0451-4000-b000-000000000000
        Vendor specific
[NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029/char002a
        f000ffc1-0451-4000-b000-000000000000
        Vendor specific
[NEW] Descriptor
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029/char002a/desc002c
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Descriptor
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029/char002a/desc002d
        00002901-0000-1000-8000-00805f9b34fb
        Characteristic User Description
[NEW] Characteristic
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029/char002e
        f000ffc2-0451-4000-b000-000000000000
        Vendor specific
[NEW] Descriptor
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029/char002e/desc0030
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Descriptor
        /org/bluez/hci0/dev_20_20_04_17_2A_C1/service0029/char002e/desc0031
        00002901-0000-1000-8000-00805f9b34fb
        Characteristic User Description
[CHG] Device 20:20:04:17:2A:C1 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 20:20:04:17:2A:C1 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 20:20:04:17:2A:C1 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device 20:20:04:17:2A:C1 UUIDs: 00001812-0000-1000-8000-00805f9b34fb
[CHG] Device 20:20:04:17:2A:C1 UUIDs: 91680001-1111-6666-8888-0123456789ab
[CHG] Device 20:20:04:17:2A:C1 UUIDs: f000ffc0-0451-4000-b000-000000000000
[CHG] Device 20:20:04:17:2A:C1 ServicesResolved: yes
[PG-9099]# trust 20:20:04:17:2A:C1
[CHG] Device 20:20:04:17:2A:C1 Trusted: yes
Changing 20:20:04:17:2A:C1 trust succeeded
[PG-9099]# connect 20:20:04:17:2A:C1
Attempting to connect to 20:20:04:17:2A:C1
Connection successful

so this says it has paired but when i go back to the emulation station and try config there is no controler detected but if i look in forget a bluetooth device it is listed there
please help 
many thanks 
david

dswann2, 2022-07-09:
no one want to tackle this?

7Ji, 2022-07-09:
This needs a specific joystick I don't have to reproduce and I doubt if other devs also have that specific joystick. But can you try if the latest test build already fixed in there?
https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.6-TEST-07082022

jemmycalak, 2022-07-15:
I have same issue with ipega sw023, device successfully connected but not detected by new controller menu.. I will try with new version.

dswann2, 2022-07-15:
tryied the new test version still same issue but i did see this on coreelec site https://discourse.coreelec.org/t/bluetooth-controllers-are-not-working/858 and as i understand it emuelec is built from this version but cant work out how this would work in emuelec

jemmycalak, 2022-07-17:
Hmmm, new version still not work. 🥲

dswann2, 2022-07-17:
ok so i can pair it with the manual pair  but when i check cat /proc/bus/input/devices
there no listing for it

Langerz82, 2023-06-03:
Please post in this thread if you have any Bluetooth Controller pairing issues:
https://github.com/EmuELEC/EmuELEC/issues/1194

closed 2023-06-03 by Langerz82